tretyakovsa/Sonoff_WiFi_switch #65: Не прошивается SonOff Wi-Fi реле.

i81, 2017-11-15:
Друзья, прошу помощи. Купил у Китайцев ещё релюшек, пытаюсь прошить - не получается:
1. Скачиваю файл ESP8266Flasher.zip, запускаю прошивальщик. В Advanced раздел SPI Mode = DIO по умолчанию. Шью. Прошивка проходит, SonOff грузится в лоадер, настраиваю Wi-Fi - к сети подключается, но WEB не работает. При этом судя по ping само устройство работает крайне не стабильно - 50% потерь пакетов.
2. Нашёл старый файл ESP8266Flasher.zip (качал в 7-м месяце и шил первое реле) запустил прошивальщик, а там по умолчанию SPI Mode = QIO. Поставил в новом (старом тоже) QIO прошил - явно лучше. Даже иногда загружается WEB интерфейс. Но проблемы те же - устройство вываливается из сети, так же 50/50 открывается web, но выглядит он по другому, не так, как мой дома (обновления через восклицательный знак ставлю там). Так же 50/50 отрабатывают http запросы управления реле.
Друзья, помогите, как решить? В чём беда?

i81, 2017-11-15:
Так выглядит плата
[image]
[image]

Так выглядит секция Advanced прошивальщика
[image]

Это web интерфейс и ping
[image]

i81, 2017-11-15:
Прошил прошивку старую - ping нормально, но WEB не открывается и /update так же

```
Сброшено подключение к веб-сайту.
Код ошибки: INET_E_DOWNLOAD_FAILURE
```

renat2985, 2017-11-15:
Давайте посмотрим какая у вас там flash память стоит.
[image]

i81, 2017-11-15:
А как это сделать? Что за программа у Вас на скрине?

i81, 2017-11-15:
ОЙ! Просвет появился!
Скачал я очередную версию архива по ссылке https://github.com/tretyakovsa/Sonoff_Loader/raw/master/ESP8266Flasher.zip
Прошил вот такими параметрами (скорость порта изменил в конфигурации самого порта в системе тоже)
[image]

Долгое время WEB не грузился полностью
[image]
Я забросил на минут 30, после смотрю - загрузился привычный интерфейс и даже кнопка управления релле работает. Потыкал на радостях релле пару раз - и опять затуп 
[image]

i81, 2017-11-15:
Ренат,  мы с Сергеем в телеграмме списались - он взялся помочь!

renat2985, 2017-11-15:
@i81 я знаю. :)

massed24, 2017-11-16:
Один в один такая же ситуация как у i85, но я прошиваю esp8266-01, купил 10штук, в магазине написано что обновленные, флешь 1Mb, в мониторе com порта,  ide size: 1048576.
Имел одну esp8266-01, покупал в другом магазине, прошил нормально, работает теперь в люстре.
Если решили проблему, поделитесь, пожалуйста, или опишите причину.
Спасибо.

i81, 2017-11-16:
Проблема пока зависела немного. Вчера остановились на подозрении про непропай т.к. работает с штырьками от прошивальщика в гнёздах, стоит их вынуть - всё сразу глюки- 90% не стабильная работа. Вот в таком виде работает
[image]
[image]

Решили прогреть плату, но до паяльной станции я не дошёл, а паяльникоп локально в районе гнёзд не помогло.
Фактически проблема не решена.

massed24, 2017-11-16:
Странно, ведь у вас, судя по фото, от прошивальщика идет TX, RX, и GND, значит устройство при прошивке питается своим стабилизатором, если бы питалось от прошивальщика,- и с ним работает, то можно предположить что не достаточно напряжения внутреннего стабилизатора.
У мена прошивается нормально, при первой конфигурации выдает IP, перезагружается, и WEB как у вас, если посмотреть 192.168.1.113/config.save.json, то там, вот такая краказябра:
� � �h�?������������ip�192.168.1 A���& Œ�,:�A��� ����0�� � ������ ��� �E���  ��� ��� q�� m�P� � '��7� P  ��9��HTTP/1.1 200 OK
Content-T
В нормально работатающем 192.168.1.95/config.save.json, вот это:
{"lang":"ru","setIndex":"index3.htm","SSDP":"Бра","space":"Комната","ssid":"???????","ssidPass":"????????","wifiConnect":"10","wifiBlink":"","checkboxIP":"1","ip":"192.168.1.95","subnet":"255.255.255.0","getway":"192.168.1.254","dns":"","timeZone":2,"ssidAP":"Sonoff-Start","ssidApPass":"","configs":"sonoff-rf","ddns":"","ddnsName":"","ddnsPort":8080,"mqttServer":"","mqttPort":15535,"mqttUser":"","mqttPass":"","time":"12:55:37","mac":"5C:CF:7F:F6:5C:55","spiffsData":"spiffs.0xBB000_flash_size_1Mb.256Kb_2017.10.16.bin","buildData":"build.0x00000_flash_size_1Mb.256Kb_2017.10.16.bin"}

i81, 2017-11-17:
Друзья, докладываю: к сожалению прогрев станцией ничего не принёс. Модуль не подключается к Wi-Fi точке доступа, создаёт свою "Sonoff-Start" - именно это имя сети прописано по умолчанию в конфигурации - но подключится к ней не получается. Пока програматор я не подключал к нему.
В момент, когда мы с Сергеем пришли к выводу, что проблема в плохой пайке или микротрещине, я решил прошить ещё один модуль с той же партии. Этот модуль пару дней у меня в холостую был включен и работал, но сегодня утром я увидел, что его нет в сети - он сам отвалился от неё. И сейчас с ним ровно то же самое, что и с первым...

DSmile75, 2017-11-19:
Мне кажется, или у вас конденсатор на плате надутый? (бочёнок по середине платы с крестом сверху) Если так, то проверьте напряжение с блока питания... И замените конденсатор.

i81, 2017-11-20:
От прошивальщика питая, я одну вообще спали :( подавал оба питания, лошарик блин...
С ёмкостью, визуально, всё в порядке, но сегодня попробую перепаять.
[image]

DSmile75, 2017-11-20:
Проверьте напряжение на соответствие в указанных точках. ESP8266 очень чувствителен к напряжению питания. Если что не так WIFI нормально не работает.
[image]

i81, 2017-11-21:
Напряжение 5,15 и 3,32 стабильно. 
Именно так и спалил прошивальщиком. В дальнейшем шился без +5В от прошивальщика - просто + не подключал, а питал от 220В.

i81, 2017-11-21:
Продолжение эпопеи: полежал этот модуль (назовём его №2 - тот, который работал только с подключенными и немного придавленными штырьками от програматора) после моей проверки, после прогрева 4 дня. Сейчас появилось время - включил проверить питание, как советовал уважаемый DSmile75 ну и попробовал в очередной раз проверить. Подключился к Sonoff-Start, настроил Wi-Fi - всё корректно прошло, модуль даже IP новый сообщил. Сейчас в сети основной и всё вроде работает... 
Оставлю включенный, посмотрим как на долго хватит...

massed24, 2017-11-21:
Может, стоит попробовать припаять керамические конденсаторы 100nF (маркировка 104) по питанию 3,3V и 5V, как дополнительные, ведь на плате стоит импульсный источник питания, поймешь, сколько он там помех гонит, и потом прошить заново, поскольку работает, но нестабильно.

i81, 2017-11-21:
А куда их паять? Прямо с зади на плату, где и менял напряжение?

massed24, 2017-11-21:
Проще именно так, там где желтыми стрелочкам на плате отмечено, там удобнее, если смд кондюки.

i81, 2017-11-24:
Не помогло
[image]

DSmile75, 2017-11-27:
[image]
 Я бы еще попробовал удалить этот конденсатор.

renat2985, 2017-11-27:
@i81 вы столько времени потратили на эту железку. Может проще купить новый sonoff? А этот в мусорник. :)

i81, 2017-11-28:
Ренат, я боюсь, что дело не в железка именно поэтому и вожусь... Смотрите: я купил 4 штуки из них две прошил - они одинаково глючат, две не трогал - работают... с такой статистикой, Вы уверены, что дело в браке железки?

renat2985, 2017-11-28:
@i81 попробуйте в таком случае более старую сборку, возможно какая-то проблема в нехватке места на Sonoff. https://github.com/tretyakovsa/Sonoff_WiFi_switch/files/1293046/Build.Spiffs.flash.1Mb.256Kb.2017.09.11.zip - здесь нет сценариев.

i81, 2017-11-28:
Ренат, ёё так шить? 
[image]

massed24, 2017-11-28:
У меня такая же картина с ESP8266- 01, купил 10 штук, пробовал прошить четыре из них и облом. Прошивается стандартно, подключаюсь к точке доступа sonoff-start, захожу по ип: 192.168.4.1, прописываю сеть, пароль, сохраняю, щелкаю перезагрузить, получаю ип: 192.168.1.111, захожу, а картина вот такая.
[image]

massed24, 2017-11-28:
Захожу по 192.168.1.111/page.htm?setup, и вижу что устройство undefined.
[image]

massed24, 2017-11-28:
А если зайти в 192.168.1.111/edit, то файловая система как бы присутствует, но в файле trigger.rgb.json и config.save.json какой то мусор как на скрине ниже.
Пробовал прошивку по ссылке, что выложил Ренат выше,- тоже самое.
[image]

i81, 2017-11-28:
Хорошо, значит определились, что я не один такой и это не физика, по крайней мере не брак...

renat2985, 2017-11-28:
@i81 да, прошиваете правильно. Все так же?
@massed24, попробуйте как и @i81 прошить более старую версию. Ссылка чуть выше.

massed24, 2017-11-28:
[image]
Ренат, я пробовал.
[image]

Вот еще скрины с Checkflashconfig, и c Termite до прошивки конечно.

i81, 2017-11-28:
Ренат,  я ещё не шил - програматор в офисе.
Ренат, но ведь @massed24 в крайнем своём сообщении написал: "Пробовал прошивку по ссылке, что выложил Ренат выше,- тоже самое." - я думаю - он это именно про "старую версию" пишет...

massed24, 2017-11-28:
i81. Да, я про старую писал, что выше

massed24, 2017-11-28:
Я в программировании не разбираюсь, но есп не одну прошил, одна esp8266-01(с первой покупки) трудится люстре, другая esp8266 -12F ролетой управляет, все норм. Складывается впечатление, что китайцы стали какую то flash запаивают дешевую, что не дружит с нашей прошивкой, что посоветует еще Ренат?

renat2985, 2017-11-28:
Я работаю только с ESP8266 07 версией. Они намного лучше других ловят роутер. Если установить дополнительную антенну можно поймать роутер через 3 стены. А вот Sonoff у меня с трудом ловит находясь сразу за стенкой от роутера, поэтому стараюсь их не использовать.
Прошил уже более 30 ESP. Все отлично, никаких нареканий. Постоянно включено около 10ESP это и жалюзи и реле и RGB. 
Но в последней нашей прошивке действительно есть какие-то проблемы с файловой системой. Если я пытаюсь загрузить какой-то файл через /edit очень часто он по какой-то причине не закачивается и ломает старый. Из за чего это происходит пока не понятно.
Попробуйте более старую версию: https://github.com/tretyakovsa/Sonoff_WiFi_switch/issues/65#issuecomment-347420106 Если она запустится скорее всего проблема с местом на ESP

massed24, 2017-11-28:
Да Ренат, я тоже пробовал загрузить файл, чтобы исправить испорченный через/ edit, вообще глюки, и зависло все, но я не уверен что корректно загружал, где то видел видео как вы это делаете не помню, не могу найти, и последнюю прошивал и версию ту, которую вы ссылку указали выше, что без сценариев(2017.09.11),- та же картина.
На последней версии даже другие устройства в сети видит, и по облаку обновляется,- но само оно undefined.

renat2985, 2017-11-28:
@massed24 пишите мне в Skype: renat2985 будем разбираться.

i81, 2017-11-28:
Друзья, а вы же потом расскажите? :)

massed24, 2017-11-28:
Обязательно, мы уже пробовали разобраться, но пока  безуспешно, есть подозрение на нехватку памяти во флешь, не сохраняется config.save.json файл, после первой конфигурации. Ренат обещал подумать, посоветоваться, отложили до завтра.

ra9mls, 2017-11-28:
Тоже сейчас ковыряюсь с прошивкой, у меня не подключается к моей сети. Постоянно транслирует свою точку.

massed24, 2017-11-29:
И так друзья, пробовали сегодня мы, значит Ренат, Сергей и я, побороть мою ESP8266 -01, фото ниже, у меня флешь P25Q80H, по даташит 1MB, какая у вас смотрите сами. В двух словах,- причина в том, что (в моей ситуации) прошивается без проблем, создает точку доступа, но при любом сохранении и после перезагрузки, повреждаются или вообще исчезают некоторые файлы системы. Это config.save.json, trigger.rgb.json и.т.д. Пришли к мнению, что это некачественная, дешевая флешь, которая не справляется с толь весомой системой.
Как только найду подходящий флешь чип для замены, перепаяю и выложу результаты
[image]

i81, 2017-11-29:
Я вначале топика выкладывал фотки своего девайса - у меня память распаяна на другой стороне платы. Моя BN25F08, если я ничего не путаю...
Вопрос: а можно сделать прошивку меньшего объема? Или сдампить оригинальную прошивку?

i81, 2017-11-29:
[image]

[image]

[image]
Слазил на люстру, посмотрел на рабочие: вроде памяти одинаковые, но конденсатора нет, который выше удалить советовали... завтра выпаяюи проверю без него...
[image]

renat2985, 2017-11-29:
@i81 мы с трудом влазим в 1Мб. Мы сейчас и не можем выпустить новую версию т.к. у нас не хватает 1 мб. А вы хотите еще в меньшей объем запихнуть нашу прошивку.... :)

massed24, 2017-11-30:
Спасибо Ренат, наверно подойдут это 4MB, но я наверно буду брать здесь: https://ru.aliexpress.com/item/Gratis-Pengiriman-10-teile-los-W25Q32BVSSIG-W25Q32-W25Q32F-25q32-W25Q32B-W25Q32FVSIG-25Q32BVSIG-ic/32696891731.html?spm=a2g0s.8937460.0.0.Q0swqO

massed24, 2017-11-30:
И так продолжение следует.
Нашел я у себя чип MX25L3206EM2I-12G, http://pdf1.alldatasheet.com/datasheet-pdf/view/934242/MCNIX/MX25L3206EM2I-12G.html, это 4МВ флешь от сат приемника, может работать только Dual Output, перепаял на место той, что на первом моем фото и теперь выглядит так.
[image]

massed24, 2017-11-30:
Прошивал вот так
[image]

massed24, 2017-11-30:
[image]

massed24, 2017-11-30:
И вот результат, устройство Ролета4, все сохраняется и обновляется по облаку, будем тестировать, напрашивается вывод,- нашей прошивке уже  мало места для корректной работу на 1МВ.
[image]

massed24, 2017-11-30:
[image]

massed24, 2017-11-30:
Всем удачи, спасибо за содействие.

massed24, 2017-11-30:
Да, не обязательно искать флешь такую как у меня, заказывайте по той ссылке, что выше выложил Ренат, не ошибетесь, важно, чтобы 4МВ было и те могут работать и в DIO

renat2985, 2017-11-30:
@massed24 не не не, не путайте людей. У меня есть несколько ESP8266 01, с 1Mb flash, они отлично работают. Скорее у вас flash с браком был. :)

massed24, 2017-11-30:
Я к тому, что если кто принял решение менять флешь, определив, что проблемы из-за неё, то уж лучше менять на 4МВ, ведь это не помешает. Да, выглядело, будто я вношу путаницу, пардон. :)

i81, 2017-12-01:
Ренат, а как же может быть, что моя флеш с браком, но при этом работает на стоковой прошивке SonOff?

i81, 2017-12-01:
Или мой вариант не то, что massed24? 
П.с. сеголня постараюсь заняться: прошить и выпяють лишний кондёр

renat2985, 2017-12-01:
@i81 у @massed24 была похожа проблема, более маленькие скетчи закачиваются и работают как нужно. Но ноша отказывается. Может у вас там просто менее 1Мб. Попробуйте перепаять Flash.

i81, 2017-12-01:
Выпаял - не помогло :( Включилась разок, чуток поработало и всё опять снова...

i81, 2017-12-01:
Перешил прошивкой Build.Spiffs.flash.1Mb.256Kb.2017.09.11.zip (которая без сценариев). Запустилась и пока работает... 
Будем подождать наверное немного...

i81, 2017-12-01:
Ренат, подскажите, что мне, далёкому от электроники, спросить у девочек в радиомагазине, что бы они мне дали нужную мне память на замену? Какими параметрами её описать?

[23 more replies]