tretyakovsa/Sonoff_WiFi_switch #66: Нет поддержки родного датчика  Si7021 от ITEAD на SONOFF TH

zalessky, 2017-11-17:
Датчик куплен тут: https://ru.aliexpress.com/item/New-Arrivals-Sonoff-Sensor-Si7021-Temperature-Humidity-Sensor-Probe-High-Accuracy-Monitor-Module-for-Sonoff-TH10/32822850571.html - вроде бы официальный магазин. Пробовал не только вашу прошивку, но и все, какие нашел альтернативные. Ни с одной прошивкой датчик не завелся. Вот что нарыл:
https://github.com/arendst/Sonoff-Tasmota/issues/735
Получается, что распайка датчика такова, что он заведется только на родной прошивке? Или что просто что-то не так делаю?
И еще вопрос: как откатиться на стоковую прошивку? Хочу-таки проверить работоспособность датчика. Вдруг с поддержкой все хорошо, просто он дохлый? Сток прошивки есть на 4pda, но не заводится девайс после прошивки. Ни в реживе DIO, ни в DOUT

Paly4, 2017-11-17:
сайт производителя - https://www.itead.cc/ru/  , я там такого датчика не увидел
с сонофом продаются датчики AM2301 и 18b20
[image]

а ваш датчик должен работать на прошивке от wifi-iot.com
на родную тоже пробовал откатится - не вышло

zalessky, 2017-11-20:
https://www.itead.cc/wiki/Sonoff_Sensor_Si7021  - он там есть
изначально Si7021 работает по протоколу i2c, то есть подключается четырьмя проводами. в этом же датчике распаяно лишь 3. 2 питания и один - на данные. Полагаю, что ITEAD намутили собственную проприетарщину. Короче, без стоковой прошивки он бесполезен

Paly4, 2017-11-20:
почему бесполезен ? я выше написал на какой прошивке он будет работать

zalessky, 2017-11-20:
Да потому что тот факт, что там заявлена поддержка - не гарантирует работу этого датчика по проприетарному протоколу. В ESP Easy он тоже [заявлен](https://www.letscontrolit.com/wiki/index.php/SI7021/HTU21D). Но не видится. 
А автор espurna говорит, что надо сниффить данные и заниматься реверсинжинирингом https://bitbucket.org/xoseperez/espurna/issues/216/add-support-for-si7021

И после этого я что-то не готов за деньги собирать прошивку от wifi-iot.com, чтобы через 5 минут снести ее нафиг.

Paly4, 2017-11-20:
фото в разобранном виде можно ?

zalessky, 2017-11-20:
Вот тут чувак припаял четвертый провод и у него завелось. вопрос: как и куда он паялся :) https://github.com/arendst/Sonoff-Tasmota/issues/735#issuecomment-323326064

zalessky, 2017-11-20:
тут схема есть https://www.itead.cc/wiki/images/7/7c/Sonoff_Si7021_Sensor_Schematic.pdf

Paly4, 2017-11-20:
я бы подпаялся к R1 и R2 к тем контактам что идут на микросхему 
[image]

zalessky, 2017-11-20:
припаялся. SDA на GPIO4, SCL - на GPIO14
На ESP Easy температура и влажность по нулям. При сканировании I2C вот какая штука получается:

`I2C Addresses in use	Supported devices
Unknown error at address 0x1
Unknown error at address 0x2
Unknown error at address 0x3
Unknown error at address 0x4
Unknown error at address 0x5
0xb	
0x24	PCF8574
MCP23017
LCD
PN532
Unknown error at address 0x25
Unknown error at address 0x26
Unknown error at address 0x27
Unknown error at address 0x28
Unknown error at address 0x29
Unknown error at address 0x2a
Unknown error at address 0x2b
Unknown error at address 0x2c
Unknown error at address 0x2d
Unknown error at address 0x2e
0x4a	PCF8591
ADS1115
LM75A
Unknown error at address 0x4b
Unknown error at address 0x4c
Unknown error at address 0x51
Unknown error at address 0x52
Unknown error at address 0x53
Unknown error at address 0x54
Unknown error at address 0x55
Unknown error at address 0x56
Unknown error at address 0x57
Unknown error at address 0x58
0x5c	DHT12
AM2320
BH1750
MPR121
Unknown error at address 0x5d
Unknown error at address 0x5e
Unknown error at address 0x5f
Unknown error at address 0x60
Unknown error at address 0x61
Unknown error at address 0x62
Unknown error at address 0x63
Unknown error at address 0x64
Unknown error at address 0x65
Unknown error at address 0x66
Unknown error at address 0x67
Unknown error at address 0x68
Unknown error at address 0x69
Unknown error at address 0x6a
Unknown error at address 0x6b
Unknown error at address 0x6c
Unknown error at address 0x6d
Unknown error at address 0x6e
Unknown error at address 0x6f
Unknown error at address 0x70
Unknown error at address 0x71
Unknown error at address 0x72
Unknown error at address 0x73
Unknown error at address 0x74
Unknown error at address 0x75
Unknown error at address 0x76
Unknown error at address 0x77
Unknown error at address 0x78
Unknown error at address 0x79
Unknown error at address 0x7a
Unknown error at address 0x7b
Unknown error at address 0x7c
Unknown error at address 0x7d
Unknown error at address 0x7e
Unknown error at address 0x7f
`

tretyakovsa, 2017-11-20:
Этот датчик фактически не является Si7021. Хотя внутри и установлен Si7021 далее все это подключено к микроконтроллеру и уже затем по трем проводам подключается к sonoff. Для включения в таком виде нужно знать протокол  обмена этого датчика. Если будет известен протокол мы подключим поддержку этого датчика.

zalessky, 2017-11-23:
Вобщем, завел я датчик. Мало припаяться к резисторам. надо еще сдуть микросхему. 
Тут с картинками шпаргалка: https://github.com/arendst/Sonoff-Tasmota/issues/735#issuecomment-346156984
ESP Easy датчик увидела. Пробую другие прошивки...

zalessky, 2017-11-23:
Хех. И сразу вопрос: в настройках Si7021 появляется только в режиме manual. И для него можно прописать лишь одну ногу GPIO - по умолчанию - 14. Как сюда добавить четвертую?

zalessky, 2017-12-17:
Я так и не смог справиться с добавлением датчика. Прошу помощи!

tretyakovsa, 2018-07-28:
Пробуйте. Датчик сделали. Проверять нечем. Подключайте к GPIO4 и 5. Датчик DHT в таком случае использовать нельзя.

tretyakovsa, 2018-07-28:
Как поставить бета версию  смотрите здесь: https://www.youtube.com/watch?v=UfikKsyVL8k&list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi&index=2