tretyakovsa/Sonoff_WiFi_switch #67: Возможно ли добавить возможность работы через ботов Telegram?

ilya-fedin, 2017-11-24:
Здравствуйте, думаю, было бы очень хорошей идеей добавить возможность управлением устройства с помощью ботов Telegram. Это дает возможность управлением устройством из любой точки мира, даже, если у тебя серый IP.
PS: также очень хочется изменить поведение лампочки, а также не хватает mqtt-**сервера** (для возможности удобного управления с помощью Mqtt IoT и других приложений, без центра умного дома в виде Domoticz и т.п.) :)

renat2985, 2017-11-24:
Управление через Telegram ботов, возможно в связке с ifttt.com. Об этом я рассказывал в видео: 
https://www.youtube.com/watch?v=v2VO3d31JRE&list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi&index=12

MQTT модуль ведь есть в прошивке. Управлять можно через https://play.google.com/store/apps/details?id=ru.esp8266.iotmanager

ilya-fedin, 2017-11-24:
@renat2985 нет, мне хотелось бы именно напрямую, без всяких дополнительных сервисов, как ifttt.
А MQTT вроде же только в виде клиента? Или таки там сервер, можно без подключения к брокеру использовать?

Paly4, 2017-11-29:
MQTT как то не полноценно или не правильно реализовано

ilya-fedin, 2017-11-29:
@Paly4 в смысле?

Paly4, 2017-11-29:
ну посмотрите что он шлет

ilya-fedin, 2017-12-03:
@Paly4 лучше вы покажите. :)