tretyakovsa/Sonoff_WiFi_switch #68: Sonoff 4CH и  Sonoff Dual

konstantin999, 2017-12-03:
Добрый день.
Спасибо за отличную прошивку! 
Интересуюсь, будет ли работать прошивка на  Sonoff 4CH и Sonoff Dual?

renat2985, 2017-12-03:
Будет, но управлять пока сможете только одним реле. В скором будущем добавим возможность управления до 8 реле.

konstantin999, 2017-12-07:
Круто! Буду ждать. Есть шансы ускорить эту разработку донатом?

Igor122017, 2017-12-09:
Здравствуйте Ренат. у меня проблема, подскажите как лучше поступить, у меня 2 выключателя sonoff wifi-rf? на одну зону, и на 2 зоны. на 1 зону я прошил вашей программой и все работает хорошо, а вот на 2 зоны, я прошил, но там оказался только 1 выключатель... в итоге, я остался без света. помогите или подскажите, как лучше поступить, чтобы заработал выключатель на 2 зоны. заранее спасибо

Igor122017, 2017-12-09:
PS, выключатель sonof на 2 зоны, стоит чип 8285

tretyakovsa, 2017-12-09:
@Igor122017 
Ссылку дайте на устройство.

Igor122017, 2017-12-09:
может лучше фото платы? будет более понятно

Igor122017, 2017-12-09:
https://www.itead.cc/smart-home/sonoff-t1.html  вот этот выключатель, только у меня на 2 зоны...

Igor122017, 2017-12-11:
Добрый вечер Ренат, у вас есть прошивка для выключателя soniff T1 на 2 зоны?

tretyakovsa, 2017-12-12:
@Igor122017 
Нет у нас такой поддержки  sonoff T1. Могу написать тестовый скетч вы его загрузите в свой модуль и  протестируете. Пришлете мне данные и я добавлю в нашу прошивку.

Igor122017, 2017-12-12:
хорошо,  я готов протестировать и предоставлю вам нужные данные

Igor122017, 2017-12-12:
и еще , может поможет, на сколько я понял: 1 выключатель сидит на GPIO5 и второй на GPIO9

Igor122017, 2017-12-12:
да, и заранее спасибо за помощь... когда примерно можно будет сделать тест? вы сможете здесь ссылку сбросить, чтобы не искать нужную прошивку....