tretyakovsa/Sonoff_WiFi_switch #69: Новые дизайны

alexandersciv, 2017-12-12:
В крайней версии прошивки предоставляются возможности переключения между 4 оформлением и интерфейсов. Default, Inline, Block, Life. Ниже скрины как они отражаются на моём мониторе в Опере.
[image]
Из них более информативный **Inline**  (все заключения моя точка зрения), но в нём тоже много пространства которое можно заполнить, на моих скринах выше подключено всего 3 датчика и уже 2 страницы. Предлагаю рассмотреть несколько набросок дизайна. Давайте обсудим.
[image] [image]
[image]

renat2985, 2017-12-13:
В ваших вариантах некоторые устройства отображаются списком некоторые в отдельных блоках. Так же и с сенсорами, некоторые просто цифрами некоторые графиками. 
Вопрос, как системе знать какие устройства выводить в список а какие в блоки, так же и с сенсорами.

alexandersciv, 2017-12-13:
**@renat2985**  При тестировании понимаю, что в период загрузки интерфейса, достаточно информации как в первой картинке. 
Слева - В главном окне - список всех устройств (увеличивается вниз)
Справа - Все Данные по устройству с которое первое по списку.
При нажатии-переключении следующего по списку устройства, справа выводятся полные его данные. В целом  этого достаточно, поскольку остальное можно получить в других интерфейсах.

renat2985, 2017-12-13:
То есть вы хотите всегда оставаться на главной странице? Если рассматривать первый ваш вариант\картинку. Когда нажимаем "Управление", "Конфигурация", вы хотите открывать не новую страницу а эти страницы подгружать на главную страницу в блок в правой части экрана, так?

alexandersciv, 2017-12-13:
Несколько не так.
На первой странице только перечень устройств со статусом + полное состояние первого, активного устройства.
Когда нажимаем "Управление", "Конфигурация" открывается новая страница, как это и реализовано уже. Закрыв страницу возвращаемся на первую.

alexandersciv, 2017-12-13:
В своё время Сергей размещал тему по Вашей прошивке https://www.ab-log.ru/forum/viewtopic.php?f=1&t=1352 там в конце пользователь **foolhome** изложил своё видение интерфейса, это к тому, что не только я вижу решения.

grossbel, 2017-12-22:
alexandersciv, если мне не изменяет память я там предлагал делать ссылки на другие устройства, поковыряв эту прошивку я понял возможности интерфейса и предложил рабочий прототип интерфейса для всей системы здесь https://esp8266.ru/forum/threads/proshivka-sonoff.1891/page-21  не знаю попалось это на глаза Ренату или нет, у самого пока не было времени этим всем больше заниматься.

alexandersciv, 2017-12-23:
**@grossbel**  Развернул предложенный по ссылке New design v2 Ваш интерфейс. Красиво и информативно, но на маленьких экранах, при наличии нескольких модулей считаю удобнее управлять по списку
[image]
а при  нажатии на управление, или общий интерфейс, или и т.д., получать всю развёрнутую картину.