tretyakovsa/Sonoff_WiFi_switch #70: Ошибки! Небольшой багрепорт

xareri, 2017-12-13:
1. При создании двух таймеров удаление первого приводит к удалению второго!
2. **Ужасный дребезг.**.. непонятно чего и тем более не понятно, почему. 
2.1. На таймере один раз реле щелкнуло и не включилось.
2.2. В веб-браузере, как desktop firefox - так и мобильном chrome - нет четкой сработки кнопки. переключение состояния явно зависит от положения звезд или от random()... после такого неверно отображается состояние реле, нужно обновлять страницу.
3. Нет версионности - я только по имени файла, который скачанный лежит на компе, могу сказать, что это сборка build.0x00000_flash_size_1Mb.256Kb_2017.10.11
spiffs.0xBB000_flash_size_1Mb.256Kb_2017.10.11
в самом интерфейсе нигде об этом не упоминается.
4. Светодиод (зеленый) светится постоянно. Было бы классно, если бы он в выключенном состоянии реле изредка кратковременно загорался сигнализируя о связи по wifi и при включеном состоянии изредка кратковременно потухал (инверсия). При поиске роутера пусть бы быстро мигал (по аналогии с индикаторами bluetooth гарнитур, что-ли...)
5. В блоке "Динамический DNS" грамматические ошибки: 
"устройство переходитЬ по ссылке." - мягкий знак здесь явно лишний. 
"динамическом IP. Что позволит" - здесь или перед "что" поставить запятую, или заменить на "Это".
"Посмотрите видео о настройкИ" - правильно "о настройкЕ"
В остальном же - я вами восхищаюсь!

renat2985, 2017-12-14:
1. Под каким образуемом такая проблема замечена? Попробуйте другой. Если под другим так же попробуйте обновить прошивку и файловую систему. - Жду ответа.
2. Не сталкивался с подобным. Не знаю чем вам помочь, это вам скорее к Сергею.
3. После того как вы установили нашу прошивку в свое устройство рекомендуется зайди на страницу "Конфигурация", "Настройки разработчика", и там нажать "Upgrade Build & Spiffs" и обновится на последнюю версию. После этого вы будите видеть какая у вас версия - она будет выделятся жирным.
[image]
5. Исправил, спасибо. Появится в новой версии.
6. Подумаю об этом, хорошая мысль.

xareri, 2017-12-14:
Приветствую!
Начал с п3. В моем случае ничего жирным не выделилось. Ну я взял и обновил последнюю Build & Spiffs (flash 1Mb 256Kb) 2017.10.16. После перезагрузки всё корректно подсвечивается и....
Коректно переключается (п.2)!
Здесь, на заглавной странице, получается, ссылка на немного устаревшую версию, вот я на неё и попался :)
[image]

п.1. Таймеры. У меня реле. Я старался со всех сторон протестировать - если записей несколько и при удалении любой, не последней - удаляется всегда последняя строка, а не та, на которой нажимаем кнопку "Удалить".
[image]

Вот как-то так...

renat2985, 2017-12-14:
@xareri после обновления проблема с удалением таймеров не исчезла?

xareri, 2017-12-16:
Неа, вот именно я экспериментировал как раз после обновления. Получается на самой свежей прошивке из доступных.

xareri, 2017-12-19:
Обнаружена еще одна достадная проблема:
Несколько дней реле было включено (тестировалась лампочка). В какой-то момент реле включилось (включилась лампочка) я хотел зайти в web-интерфейс. Но не смог. Пришлось реле перезагрузить путем выключения из сети.
Не знаю, софтовая ли это проблема или железная. Но после этого случая я серьезно задумался о надежности подобных вещей.

ilya89vet, 2018-02-21:
Sonoff от 2017.10.16
Проблема со сценариями ИК пульта
При добавлении сценария "Вкл/Выкл ИК пультом" пультом реле вкл. и выкл. нормально, НО через веб интерфейс или MQTT перестает работать (при нажатии кнопки веб интерфейса или MQTT, реле дергается на 1 сек). 
Только убираешь сценарии и все начинает нормально работать! Но ИК пульт конечно перестает работать.
PS. возможно этот-же баг есть и на RF пульте, но у меня нет возможности проверить.