tretyakovsa/Sonoff_WiFi_switch #71: Кирпич после штатной перепрошивки и элементарной настройки

zalessky, 2017-12-17:
Sonoff basic. Прошил на версию 2017.10.11
Настроил реакцию на PIR датчик 
Настройка времени пока не активна. Но в целом - 2 минуты по умолчанию устраивают
Припаял датчик, врезал его в потолок, реле - в стену, светильник - в нишу. Подал питание - все работает. Порадовался. Показал домашним. 

Через час обнаружил, что есть прошивка посвежее: 2017.10.16. Прошил по воздуху. Настройки сбились: галки по умолчанию стоят. Датчики  температуры-давления включены, датчик движения - выключен. Поставил как было. Ребутнул релюшку и на этом все. В сети не регистрируется, на движение не реагирует, свою WiFi сеть не поднимает :(((

Понимаю, что можно просто перепрошить по проводу и все снова заработает. Но для этого надо демонтировать назад и отпаивать датчик на весу. обидно, понимаешь...

renat2985, 2017-12-17:
https://www.youtube.com/watch?v=D0Fu8etyVg0&index=23&list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi

mikemdr, 2017-12-18:
Может, дело в том, что в новые Sonoff basic стали запаивать медленные чипы Flash-памяти. По-этому, компилировать и шить их надо с опцией "Flash Mode: DOUT" (а не DIO), т.е. третий байт в прошивке должен быть 0x03 (а не 0x02 как в прошивках, прошиваемых "по воздуху"с сайта). У меня старый модуль после обновления "по воздуху" отлично работает, а недавно купленный - после обновления виснет при операциях с флешом (обновляю по проводам со сменой 0x03 на 0x02 в прошивке с github'a).

zalessky, 2017-12-18:
Видео посмотрел ночью. Вот только ссылка куда-то делась... 
Мне сложно сказать почему так бывает. Но при обновлении по воздуху первое включение проходит штатно, но слетают все настройки. И вот после перенастройки из ребута девайс уже не выходит. 
Рецепты из видео не подходят - для того, чтобы перепрошить, надо его завести. А он кирпич. Шьется только проводом. И, да. Шью с опцией DOUT всегда - как в инструкции

Xman0674, 2018-02-13:
Аналогичная ситуация. Пробовал установить на SonOff данную прошивку  через лоадер - прошивается, перезагружаешь модуль, поднимается точка, указываешь данные wifi сети и модуль сам ребутится и скачивает в онлайне какую то (наверное самую новую) прошивку. После того как светодиод загорается зеленым и сонофка появляется в сети на него можно зайти и сделать настройки. Но после первой же перезагрузки окирпичивается. Память медленная, шьется только в DOUT . Потом нашел видеоинструкцию вашу как прошить по проводу последнюю прошивку - по проводу прошивка встала нормально и после ребутов запускается нормально. Что если перепаять чип памяти? купить жменю с али например? Ну не хочется лезть на потолок чтоб обновться в будущем.  Для люстры конечно и так прекрасно сгодится и можно не обновлять...

renat2985, 2018-02-13:
@Xman0674 причин может быть много. Все прошивки (bin файлы) компилируются под 1Mb flash size. DIO, то есть при обновление, если ранее вы указали другие параметры, скорее всего у вас ничего не получится.
Через несколько недель мы наверное все таки выложим новую версию. Попробуйте потом обновится, если не получится то покупайте новую Flash память. У нас был issues на эту тему, к сожалению не смог его найти. Человеку помогла перепайка, но у вас немного другие симптомы. 
Может быть проблема и с нашей стороны... В общем предлагаю подождать до новой версии.

Xman0674, 2018-02-14:
@renat2985 При прошивке с параметром DIO точно не работает, пробовал, сразу же не стартует или прошивальшик вовсе стопорится на середине. Новые модули SonOff идут  с медленной памятью и их можно прошивать только с DOUT . Ну мне пака что и так сгодится, главное что выяснилось из-за чего происходит.

KostyanApa, 2018-09-27:
SOnOff
Прошил с опцией DOUT, зашел на нее, прописал логин/пароль от WiFi и больше ее не видел.
Подскажите как раскирпичить?