tretyakovsa/Sonoff_WiFi_switch #72: Где храняться настройки WiFi?

mikemdr, 2017-12-19:
Залил последнии (2017.10.16)прошивку и SPIFF, подключился к WiFi на работе, настроил всё на странице "Конфигурация" - всё работает. Меняю настройки для подключения к другой точке, настройки сохраняются, передёргиваю питание Sonoff  - в "Конфигурации" новые настройки WiFi, но подключено к СТАРОЙ точке. Выключаю старую точку , надеясь, что Sonoff  перейдёт в режим АР - но после указанного в "Конфигурации" времени, Sonoff  просто перезагружается и пытается подключиться к СТАРОЙ точке. И так - по кругу. Заливаю в Sonoff  ПУСТУЮ SPIFF - Sonoff  пытается подключиться к СТАРОЙ точке... Заливаю мной компилированную прошивку в режиме "1М(64кSPIFF)" без SPIFFа - коннект к СТАРОЙ точке. Похожее (только не-убираемый режим АР) уже описывался. 
И как в этом случае менять настройки WiFi?

renat2985, 2017-12-19:
Меняются только здесь: http://192.168.1.247/page.htm?first-connect

mikemdr, 2017-12-20:
Я настраиваю Sonoff на работе, отвожу на дачу - как подключиться к Sonoff по СТАРОМУ ip?

xareri, 2017-12-20:
mikemdr при первом включении реле становтся точкой, присоединяешься к нему. Настраиваешь на сеть своей дачи (имя точки, пароль). По адресу http://айпишник_реле_в_режиме_точки(192.168.4.1_вроде)/page.htm?setup можно установить фиксированный IP. (Это IP адрес шлюза в свойствах соединения)
И если конфликтов не будет - то на даче в сети реле будет иметь указанный адрес.
Если не заморачиваться с фиксированными айпишниками - можно на даче его включить, а потом **сканером сети найти** его. **В маршрутизаторе** это тоже обычно видно. Но лучше его по МАК адресу **прибиндить**. Чтоб адрес всегда один и тот же был.

mikemdr, 2017-12-21:
В первом сообщении я сообщал, что модуль (при отсутствиив эфире уже прописанной в нём точки доступа) не переходил в режим АР, а просто перезагружался по wdt и опять искал прописанную ГДЕ_ТО  в глубине своей флеш-память точку. Это (возможно) стало проявляться после обновления по-воздуху на последнюю (от 16.10.2017) прошивку, которая не смогла корректно прописаться из-за флеша, работающего только в DOUT. 
Помогла только полная очистка памяти записью blank_1M.bin флешером.
По-этому, я и спрашивал где (eeprom, spiff, rtc) "храняться настройки WiFi" , а не с какой страницы интерфейса получить к ним доступ.

mikemdr, 2017-12-21:
В первом сообщении я сообщал, что модуль (при "отсутствии в эфире" уже прописанной в нём точки доступа) не переходил в режим АР, а просто перезагружался по wdt и опять искал прописанную ГДЕ_ТО в глубине своей флеш-памяти точку. Это (возможно) стало проявляться после обновления по-воздуху на последнюю (от 16.10.2017) прошивку, которая не смогла корректно прописаться из-за флеша, работающего только в DOUT.
Помогла только полная очистка памяти записью blank_1M.bin флешером.
По-этому, я и спрашивал где (eeprom, spiff, rtc) "храняться настройки WiFi" , а не с какой страницы интерфейса получить к ним доступ.

mikemdr, 2017-12-21:
В первом сообщении я сообщал, что модуль (при "отсутствии в эфире" уже прописанной в нём точки доступа) не переходил в режим АР, а просто перезагружался по wdt и опять искал прописанную ГДЕ_ТО в глубине своей флеш-памяти точку. Это (возможно) стало проявляться после обновления по-воздуху на последнюю (от 16.10.2017) прошивку, которая не смогла корректно прописаться из-за флеша, работающего только в DOUT.
Помогла только полная очистка памяти записью blank_1M.bin флешером.
По-этому, я и спрашивал где (eeprom, spiff, rtc) "хранятся настройки WiFi" , а не с какой страницы интерфейса получить к ним доступ.