tretyakovsa/Sonoff_WiFi_switch #73: Сценарии по температуре-влажности

zalessky, 2018-01-31:
Не смог сделать сценарий управляния вытяжным вентилятором по достижению высокого показателя влажности. Не реализовано? Или я плохо искал?

renat2985, 2018-01-31:
В разработке. Пока не реализовано.

grossbel, 2018-02-01:
Привет всем, что с проектом, застыл, умер или в глобальной переделке?

renat2985, 2018-02-01:
Все немного затянулось. Но Сергей практически каждый день что-то шаманит :) скоро выложим.

tretyakovsa, 2018-02-09:
Можно, только памяти все меньше и меньше. И на сколько это актуально для Sonoff не понятно.

Cvazirus, 2018-02-09:
А если подключать только те модули что нужны в данный момент.

Cvazirus, 2018-02-09:
Просто ребята у вас получилась отличная вещь. Надо развивать.

Svet98, 2018-02-19:
Всем привет и огромное спасибо за Ваш проект!

У меня датчики ds18b20 купленные года два назад отказываются работать на есп12е 14 порт вообще (т.е. даже на тестовой ванваре или далас), но работают отлично на 2 порту.
Хотела просто исправить в Ваших исходниках порт под себя, но сырцы которые выложены на момент написания этого сообщения, не компилируются. Можно попросить Вас порт на второй поменять? пожалуйста...

И еще вопрос.
Первый раз когда прошивала есп-01 с dht11, график работы градусника присутствовал.
Пока мучала ds18b20 (на другой есп-01), график градусника отсутствовал. В коде есть проверка на подключен или нет? Не всегда удобно, а в некоторых случаях и не возможно подключить переферию.

По прошивке на есп-01
Прошивала несколько есп-01 по проводам. Окирпичились после повторной прошивки. Пару дней не могла завести их. Может проведение, а может и реально, помогла заливка даты из скаченного скетча и поверх уже бинарники через флешер.

Спасибо)

grossbel, 2018-02-20:
Резистор подтяжки стоит на шине датчика? у меня на ESP12 работают все.

Svet98, 2018-02-20:
Конечно стоит. Я разводку сразу на датчике сделала. Без разницы в какой порт подключать, подтяжка будет.

Svet98, 2018-03-01:
Всем привет.
Сергей, когда планируете реализовать в прошивке "выбор порта" на ds18b20?
Кстати, код можно значительно уменьшить. Нашла на "просторах" и попробовала в работе.
`#include <OneWire.h>
#include <DallasTemperature.h>
#define ONE_WIRE_BUS 2

OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);

void setup(void)
{
  Serial.begin(115200);
  sensors.begin();
}

void loop(void)
{ 
// если датчиков несколько, в скобках нумеруем каждый начиная с 0. Не пробовала.
  sensors.requestTemperatures(); // Send the command to get temperatures
  Serial.println(sensors.getTempCByIndex(0));
}`

И ещё вопрос. Из за ограничений по размеру памяти у Вас не получается значительно расширить функционал, приходится многим жертвовать. Возможно ли создание нескольких прошивок, каждая из которых поддерживает свой перечень оборудования? Думаю, даже если придется открывать для каждой группы свой браузер, это будет не так страшно, как отсутствие возможности вообще использовать другое оборудование.

Svet98, 2018-03-01:
Кстати, если вы включите в свой проект отдельный управляющий веб сервер, который будет собирать данные со всех esp не взирая на прошивку, будет такое же единое поле. Сам веб сервер, можно разместить на esp с большим объемом памяти.

lalayan089, 2018-09-27:
В прогрессе уже почти год? Нам очень нужна эта функция, пожалуйста поработайте над этим, спасибо.

renat2985, 2018-09-28:
@lalayan089  Уже работает в бета версии, пробуйте: https://www.youtube.com/watch?v=UfikKsyVL8k&t=0s&index=3&list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi