tretyakovsa/Sonoff_WiFi_switch #74: Можно ли сделать больше реле?

Vopseni, 2018-03-18:
Даже самый простой модуль ESP-01 поддерживает 4 реле так как выведены GPIO0, GPIO1, GPIO2, GPIO3, можно ли управлять большем количеством реле, хотя бы 4-мя? Как это настроить?

renat2985, 2018-03-19:
Уже сделали управление до 4 реле, на нужных вам GPIO, и до 7 стационарных кнопок, так же можно настраивать на нужные GPIO. Это уже все есть в исходниках. Но скомпилировать скорее всего корректно у вас не получится. Скоро выложим bin файлы новой версии.

Vopseni, 2018-03-19:
Спасибо за ответ!
Но на самом деле последнею версию жалюзи я компилировал в среде IDE Arduino 1.6.5 и все хорошо прошло и получилось
Но последняя версия “Sonoff_WiFi_switch” в процессе компиляции выдает ошибку:” 
In file included from Sonoff_WiFi_switch.ino:1:0:
sets.h:1:25: fatal error: ESP8266WiFi.h: No such file or directory
 #include <ESP8266WiFi.h>
compilation terminated.
Ошибка компиляции.»
Отсюда вопрос, в какой версии среды нужно компилировать последнюю версию программы? 
P.s. Пробовал и в версии IDE Arduino 1.8.5  то же ошибка.

renat2985, 2018-03-20:
Таймеры в новой версии пока не работают. Сенсор движения работает, но его нужно дополнительно настраивать в вебе. Если очень нужно, пишите расскажу.

wellw, 2018-03-22:
Здравствуйте Ренат. Очень хороший у вас проэкт. респект и уважуха! 
не могли бо вы мне подсказать один ньюанс. как сделать так чтобы при управлении gpio2 при нажатии кнопки сначало появлялась логическая 1 а при отпускании 0.
в идеале на gpio2 всегда должен быть низкий уровень. только при активации высокий на мгнавение. можно это так сказать на "горячую" поменять через функцию edit ? 
без перепрошивки!?

Vopseni, 2018-03-22:
Как плохо что таймеры не работают мне они и нужны, а когда сделаете?

TimKaSANOFF, 2018-04-29:
Когда выложите прошивку с поддержкой "нескольких реле"???
А то пора уже поддержку gsm модулей заказывать...типа sim800L(C ..и.т.д.):)

Vopseni, 2018-05-03:
Подскажите, а так и не доделали работу таймеров с множеством реле? Я думаю что это очень необходимо, это так сказать минимальный функционал (очень нужен автополив, сезон начался)

renat2985, 2018-05-03:
@Vopseni, пока нет. Вы можете использовать DDNS и сервис ifttt.com для таймеров.
То есть сервис ifttt в нужные вам дни и время будет делать GET запросы на ваш ESP, тем самым включать и выключать реле. Подробней можно посмотреть здесь:
https://www.youtube.com/watch?v=v2VO3d31JRE&index=13&list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi
Порты уже работают.
Других вариантов пока нет.