tretyakovsa/Sonoff_WiFi_switch #75: Вопрос про управление Gpio из вашей прошивки

wellw, 2018-03-22:
Здравствуйте Ренат. Очень хороший у вас проэкт. респект и уважуха!
не могли бо вы мне подсказать один ньюанс. как сделать так чтобы при управлении gpio2 при нажатии кнопки сначало появлялась логическая 1 а при отпускании 0.
в идеале на gpio2 всегда должен быть низкий уровень. только при активации(нажатии кнопки) высокий на мгнавение. можно это так сказать на "горячую" поменять через функцию edit ?
без перепрошивки!? или подскажите в каком файле и что нужно редактировать. пожалуйста

renat2985, 2018-03-23:
Насколько я понимаю вам нужна инверсия. Если у вас последняя версия то есть вы ее сами компилировали с исходников. Попробуйте на странице конфигурации, модулей. В TACH после времени в мс. добавить через пробел цифру 1, далее сохранить и перезагрузить. 
[image]

Поле отпишитесь, работает или нет.

wellw, 2018-03-23:
Спасибо .попробую. Я не компилировал.  Версия того года сентября.
это в профессиональных настройках!? такого нет у меня. где взять последнюю версию?

renat2985, 2018-03-23:
Если вы скачали и загрузили bin файл прошивки в esp, у вас не будет нужной вам функции.
bin файлы новой версии выложим позже. Когда решим какие функции сделать платными. В этом вы можете помочь в созданной сегодня теме.

wellw, 2018-03-24:
За возможность  обновление и активацию определенных функций. )))
вопрос как.... Пусть тот кто хочет подержать проект может приобрести Платную прошивку с расширенным функционалом и возможностью обновления и расширения функций. А кто по каким то причинам не имеет возможности либо желания поддерживать ПРОЕКТ тот может пользоваться ОГРАНИЧЕННЫМ функционалом. Илементарно Ренат ))) И сделать три типа прошивок Premium,Basic и free. А Лицо имеющее Лицензию получает гарантированную Техническую поддержку. Плацдарм для рекламы есть это Ютуб. Вот такой вот бизнес план и не дорого)))

wellw, 2018-03-25:
Ренат. Может быть какой то ваил можно поменять чтобы TACH появился тот какой мне нужен ?
пробывал за компилировать самостоятельно последнюю прошивку 8 библиотек добавил в среду ардуино. Не помогло. ошибка вылетает . как всегда не информативная. т.е мне не понятно о чем.
а так у меня много файлов из интернета кампилируются а этот нет. версия 1.84

renat2985, 2018-03-25:
Нужная вам функция доступна только в новой версии. Которую можно собрать из исходников.
Ждите, через несколько недель выложим выложим bin файлы.

wellw, 2018-03-25:
Спасибо буду ждать

alex-xs, 2018-04-03:
добрый день! Ренат какая у вас версия IDE?

renat2985, 2018-04-03:
@alex-xs, v1.8.5