tretyakovsa/Sonoff_WiFi_switch #76: Монетицазия проекта. Как? Нужные ваши мысли.

renat2985, 2018-03-23:
## Для развития любого проекта к сожалению нужны инвестиции иначе он просто "умрет".
Подскажите за какие функции вы готовы платить?
Делать новые версии полностью закрытыми и выкладывать за деньги.
Или урезать какой-то функционал. 
Поделитесь пожалуйста своими мыслями.

wellw, 2018-03-23:
Конечно нужно поддерживать,тестировать и развивать. Бесплатно этим не кто заниматься не будет.
Ренат я вчера тебе сообщение написал. Сегодня его могу найти.... Нужна твоя помощь. Естественно не безвозмездно.

Sawdevil, 2018-03-26:
Покупка лицензий на каждую ESP отдельно или покупка новых версий. Очень не хватает  поддержки нескольких  GPIO.

Aval0ncheG, 2018-03-27:
Я вижу 2 пути:
1. Это все новые версии платные.
2. Оцениваете трудозатраты на добавление определенного функционала, выкладываете "прайс" и дальше что то на подобии краудфандинг, каждый кому интерестна та или иная функция в меру своих возможностей платит, как насобиралась необходимая сумма реализовали.

renat2985, 2018-03-28:
Если делать новые версии платными, получается нам нужно закрывать Open Source. Исходники не будут доступны людям. И у некоторых могут возникнуть параноидальные мысли, "код закрыт, а вдруг там зашили вредоносный код и за мной сейчас следят и могут управлять моим домом!". :)

Мысль о краудфандинге интересная, спасибо.
Буду рад услышать еще мысли.

Sawdevil, 2018-03-28:
Я вот в коде мало что понимаю, зашил прошивку минимум движений и все работает. Кому надо пусть купят исходники. Главное не бросайте проект. С нетерпением жду новую версию)

Svet98, 2018-04-03:
Добавлять новые функции по просьбе трудящихся за разумную плату.
Одни по жизни халявщики а другие хотели бы платить, но состояние политики любимых государст заставляет больше думать что купить, очередной девайс для души или яблоко для ребёнка.
Но решать Вам. И, спасибо за открытый проект. Для кого то просто интересно "поковырять", а у кого то банально нет возможности платить достойно.

tigr72, 2018-06-07:
Релизы только в форме бинарников и за деньги, причем с валидацией на внешнем сервере по МАС ESP-шки. Если нужны исходники конкретного релиза, тоже плати. 
Ветку *-master оставить в открытом доступе - качай, собирай, тестируй новые функции, вылавливай баги и  не плати за это...
В общем, по принципу Fedora и RedHat...

janisbvp, 2018-06-17:
Только что скачал и залил этот софтик но сразу могу сказать что стабилнее и интереснее, чем некоторые готовые/комерческие варианты.
Увы, в наши дни мало кто добровольно будет донатить проекту, хотя я надеюсь что ошибаюсь.
По любому, проект стоет поддерживать, так как на сегодня мало решений, которые каждый может заставить работать. 
Извиняюсь за ужасную граматику...