tretyakovsa/Sonoff_WiFi_switch #80: Не работает сборка на исходниках от 10.04.2018

Svet98, 2018-04-14:
Что я не так делаю?
Пробовала два варианта железа.
1. стандартный ESP01 от роботдина.
    -  Не работает таймер. Не записывает вводимые данные. Лист пуст.
    -  Стоят: RGB WS2812 на потолок в детской + DTH11
    Сделала сценарии для RGB с переходами в разные динамические режимы. В проверочной 
    строке (глазик) отрабатывает корректно. После добавления в список, динамический режим для 
    всех сценариев подставляется из настроек configs/rgb игнорируя сценарий.

В настройках RGB у меня стоит 19 режим
Сценарии для rgb такой:
--
Если Время = 07:00:00 Тогда this rgb on 00cc00 1 255 2
Если Время = 07:15:00 Тогда this rgb off
Если Время = 20:00:00 Тогда this rgb on FF0000 1 255 2
Если Время = 20:05:00 Тогда this rgb on FF0000 1 155 19
Если Время = 21:55:00 Тогда this rgb on 00FF00 1 255 2
Если Время = 22:00:00 Тогда this rgb off

В итоге, вместо динамического режима 2, везде 19.

2.   на стандартном ESP01 нет выведенного 14 и А0 портов. Под рукой оказалась D1 mini на ESP8266 от роботдина с портами D0-D8, A0.
- DTH11 запустился всего один раз и очень странно. В настройках 2 порт, на плате 3 
   порт. После многочисленных прошивок вообще отказывается появляться на любых портах.
- RGB WS2812  не работает не на одном порту. Вообще.
- Если подключить WS2812  к портам D4, D7, D8 система выходит из строя (http не работает, ping 3- 
   5 ответов, потом провалов несколько и по кругу). Оживляется только перепрошивкой Sketch 
   Data Upload.
- Датчик света TEMT6000 на А0 работает корректно.

П.С, В настройках уже давно нет датчика движения. Он появится?

Svet98, 2018-04-14:
Последний готовый 6 месяцев назад был. Или я не там смотрю?

armaros77, 2018-04-14:
Тоже собирал из исходников, которые были еще на 02.04.2018. Вчера обнаружил что таймеры не добавляются.

renat2985, 2018-04-15:
Svet98, да. Новую версию ещё не допилили

Svet98, 2018-04-15:
Ясно, спасибо.
Пока болею, могу подебагить ))

renat2985, 2018-04-16:
Svet98, компилируя исходники самостоятельно у вас может не стабильно работать прошивка. Это может быть не соответствие версий библиотек. ESP может вообще не запустится т.к. в репозитории не всегда работающий код. Поэтому лучше используйте готовые bin файлы.

Svet98, 2018-04-18:
А можно узнать версии библиотек?
Спасибо.

Svet98, 2018-04-21:
Спасибо.
Про ядро Вы говорили, ставлю именно 2.3.0
Библиотеки все по ссылкам Вашим ставлю. Но есть еще библиотеки IDE, вот про их версии я и спрашиваю. Может из-за них у меня происходят чудеса. Если не сложно, выложите их версии.

tretyakovsa, 2018-04-22:
Я обычно использую 1.6.13 для windows. Все версии можно найти здесь: https://www.arduino.cc/en/Main/OldSoftwareReleases#previous

Svet98, 2018-04-24:
Спасибо

zon2004, 2018-04-26:
Уважаемые разработчики. Подскажите пожалуйста. Перед новым годом делал гирлянду на елку на Вашей прошивке. И вроде как можно было прописать какую программу по умолчанию она запускает при подаче питания. Теперь делаю ночник в детскую, скачал заново файлы прошивки, залил, но не могу найти теперь куда можно прописать что стартует по умолчанию. И еще теперь при включении не включается даже статика. В вебе пишет включено, а на самом деле все светодиоды выключены. В вебе выключить потом включить и все работает. Подскажите пожалуйста, как хоть вернуть чтобы при подаче питания включалась хоть статика

renat2985, 2018-04-26:
@zon2004 пожалуйста создавайте новые issues! По поводу включения после рестарта: "Конфигурация" -> "Настройки разработчика" -> "Дополнительные настройки" -> "EDIT" далее ищем RGB и в последней ячейке вместо 0 ставим 1.
[image]

alex-xs, 2018-04-27:
Здраствуйте! У меня при компиляции выдает приктически по всем переменным  was not declared in this scope

closed 2018-04-27 by renat2985

alex-xs, 2018-05-02:
Вы меня извините - только учусь. Версия ide 1,6,13 , ядро esp 2.3.0, библиотеки из sets.h -ошибка та же.

Svet98, 2018-05-02:
alex-xs, скиньте лог, попробую помочь вам.

alex-xs, 2018-05-03:
[log.txt](https://github.com/tretyakovsa/Sonoff_WiFi_switch/files/1969767/log.txt)

Svet98, 2018-05-03:
Я так понимаю у Вас стоят несколько версий IDE? Это мешает нормальной работе.
Попробуйте снести IDE, удалить полностью все папки (лучше перенести в архив, мало ли) по путям (из того что вижу):
C:\Documents and Settings\aleks\Local Settings\Application Data\Arduino15
C:\Documents and Settings\aleks\Arduino\
F:\arduino-1.6.13\
помойку тоже почистите x:\Temp\

ESP ставьте не через менеджер плат а по старинке через настройки. У меня через менеджер плат компилить отказался, ява в ошибки уходила. Снесла всё, поставила через настройки и всё заработало. Ссылка как ставить https://geektimes.com/post/271754/

Есть еще одна неприятность. Старые прошивки полностью не затираются. То что Вы записываете на модуль, "ложится" по верх того что уже там есть. Отсюда глюки могут быть.
На днях, на один из модулей для экспериментов я прошила из примера тест bme280. Датчик исправно работал, но!!! Данный модуль у меня успешно пингался  ))). Проверила физическим включением выключение модуля. ))) Сейчас пойду читать как "почистить" )))

Svet98, 2018-05-03:
Алекс, предлагаю списаться на прямую, что бы не мучать нервы наших Уважаемых гуру.
Если выясним что то полезное, просто отпишем после.
Если не против, мой емайл svet740@gmail.com
И если не сложно, скиньте картинку (скрин) первых строк ошибки в самом IDE. Я по ним лучше ориентируюсь.

alex-xs, 2018-05-04:
вы светлый человек Светлана....