tretyakovsa/Sonoff_WiFi_switch #81: Неправильно отображает значение температуры и влажности датчика DHT

Aldaran, 2018-05-11:
Скачками отображает параметры температуры и влажности, то правильное значение, то "2147484000".
Сунул напрямую получение значения от датчика "dht.getHumidity()"  в строку "String data = graf(getStatusInt(humidityS), 10, t, "low:0");" вместо получения переменной getStatusInt(humidityS), проблема ушла, забыл на некоторое время, но тут решил сделать калибровку датчиков (разброс просто ужасный). В общем все даже заработало =) Пока ковырял код был один момент;
String data = graf((int)(dht.getHumidity()*getOptionsFloat(cHumiS)), 10, t, "low:0");
Тогда увидел эту же проблему. В варианте без перехода на Integer, все хорошо работает.

tretyakovsa, 2018-05-12:
Попробуйте getStatus(humidityS),  вернет строку.

Aldaran, 2018-05-16:
Использую getStatusFloat(humidityS), т.к. getStatus(string ...), а graf(float ...).

almaz8811, 2020-04-24:
Тоже скачет температура, отображает правильные показания, потом 2147484000, и так каждые две секунды. Менял библиотеки, не помогло.

tretyakovsa, 2020-04-24:
@almaz8811 Я тестировал с датчиками DHT 11 двух производителей. 
AOSONG старый работает. ASAIR новый дает описанный вами эффект. 
У Вас какой датчик и какие библиотеки вы пробовали?

almaz8811, 2020-04-24:
> @almaz8811 Я тестировал с датчиками DHT 11 двух производителей.
> AOSONG старый работает. ASAIR новый дает описанный вами эффект.
> У Вас какой датчик и какие библиотеки вы пробовали?

Скорей всего датчик у меня нового типа. Запрос dht.getMinimumSamplingPeriod() выдает 1 секунду, а по факту новому датчику нужен интервал опроса в 2 секунды. Решил проблему следующим образом:

> void initDHT() {
  uint8_t pin = readArgsInt();
  pin =  pinTest(pin, HIGH);
  dht.setup(pin);
  delay(1000);
  static uint16_t t = readArgsInt();
  static uint16_t test = dht.getMinimumSamplingPeriod() * 2;  //  TODO: Время опроса датчика умножаю на два, иначе DHT11 показывает не верные значения

Всё заработало нормально.

tretyakovsa, 2020-04-24:
@almaz8811 Не понял зачем код менять если в файле конфигурации можно написать DHT 2 2000

almaz8811, 2020-04-25:
> @almaz8811 Не понял зачем код менять если в файле конфигурации можно написать DHT 2 2000

не найду в каком конкретно файле редактировать параметр