tretyakovsa/Sonoff_WiFi_switch #85: Проблема с обработкой кнопок по сценариям.

multiminik, 2018-06-14:
Прошивка собрана из исходников по состоянию на 10.06.2018. При создании сценария:
если sonoff состояние конпки(gpio-0) = 1, тогда sonoff реле = not 1(gpio-12)
всё отлично кнопка переключает состояние реле.
При создании сценария:
если sonoff состояние конпки1(gpio-9) = 1, тогда sonoff реле = not 2(gpio-4)
кнопка1 не переключает реле2 - нет никакой реакции.
При создании обоих выше описанных сценариев отрабатывает только первый.
Если к ним добавить сценарий:
если sonoff состояние конпки2(gpio-10) = 1, тогда sonoff реле = not 3(gpio-5),
тогда при нажатии на кнопку0 отарбатывает её сценарий и попутно кратковременно включается реле3.
Подсажите что не так?

tretyakovsa, 2018-06-15:
Это обусловлено тем,, что использование GPIO6 - GPIO11 под большим вопросом и если, это происходит значит так делать нельзя. Проверка доступности пинов будет встроена.

multiminik, 2018-06-15:
В sonoff ch4 gpio 9(sdio2) и 10(sdio3) по умолчанию висят на кнопках и нормально отрабатывают на родной прошивке. Каким образом они это реализовали? Не разбирались?

tretyakovsa, 2018-06-15:
Вы когда собирали прошивку какой тип обращения к памяти устанавливали в настройках.
[image]

multiminik, 2018-06-15:
DOUT.
Модуль psf85a стоит. На нем флеш вроде выстроенный.

multiminik, 2018-06-15:
Смотрю проверку занятости пина уже добавили?

multiminik, 2018-06-17:
Пока решил свою проблему, задействовав gpio1 (TXD) и gpio3 (RXD), которые по умолчанию в вашей прошивке тоже сходу не работают (и как я понял не должны для работы с UART). Пока сделал так:
[image]
Всё прекрасно работает. Если при этом добавить невозможность включения UART на странице конфигурации, то думаю можно так и оставить. Может кому пригодится, если не нужен порт для отладки или для управления реле через него.
Так же нашёл информацию, что gpio9 и gpio10 можно использовать как дрыгалки при условии, что тип обращения к памяти DOUT. Чуть позже куплю ещё один модуль CH4, так как тот который есть - переделан и становится сегодня на службу, и буду пробовать заставить его работать на родных кнопках.
Ваша задумка изумительна! Функционал на уровне. За то, что устройства не привязаны к общему серверу отдельное спасибо!

tretyakovsa, 2018-06-20:
Я в принципе могу учесть в коде, что если режим DOUT, то эти пины можно использовать.

multiminik, 2018-06-21:
Было бы не плохо. Получилась бы полная поддержка 4ch. Плюс датчики есть куда повесить.

Jawa12, 2018-08-06:
Да , ещё, собирается только при установке менеджера плат esp8266 2.4.0-rc2 и там нет ещё возможности выбора типа обращения к памяти... В противном случае компилятор ругается на обращение к удалённой функции...

multiminik, 2018-08-07:
@Jawa12 
А если выбрать плату generic и там всё под esp12e настроить? Для generic выбор типа доступа к памяти имеется.