tretyakovsa/Sonoff_WiFi_switch #86: Локальное хранение "Starting Log" и какие ещё данные отправляются на http://backup.privet.lv

mikemdr, 2018-06-19:
Стоит версия "Последняя стабильная" (от 2017,10,16).
Возникла необходимость разобраться с неожиданными "умолканиями" модулей, установленных на даче - подозрение пало на проблемы с электричеством. 
Загружаю страницу "Starting Log" (http://[ip_sonoff]/page.htm?starting) - в "проблемные периоды" ничего нет. 
_ДОЛГО_ "копал" в других направлениях, пока случайно не заметил, что не все перезагрузки отображаются в  "Starting Log". Оказывается, что лог хранится не в Sonoff-модуле, а на "http://backup.privet.lv/visitors/{{mac}}.json". Соответственно, под "Starting" надо понимать НЕ удачный запуск модуля, а удачную отправку МОИХ данных разработчику.   
Вопрос, можно ли, не перекомпилируя прошивку:
- включить хранение логов локально;
- отключить СКРЫТУЮ слежку за пользователем.
P.S. Вообще то, принято ЗАРАНЕЕ предупреждать о таких вещах, и пользователь САМ должен решать - соглашаться на это или искать другой проект. В данном случае, (у меня) возникают "нехорошие мысли" по поводу и других моих данных (в т.ч. паролей)...
P.P.S. Ничего личного - обычная паранойя (предусмотрительность) сис.админа...

renat2985, 2018-06-19:
Без пере компиляции и удаления\добавления кода нельзя.
Опираясь на эту статистику мы можем быстро выявить проблемные версии и сделать в них фиксы.
Пароли мы не собираем, исходники проекта открыты, вы можете проверить все сами.
Отключать эту функцию не планируется. Посещая любые сайты вы куда больше отдаете своей личной информации их владельцам.
В новой версии на странице Starting log будет описано что это за информацию и где она хранится.

multiminik, 2018-06-19:
Можно сделать платную кнопку для предусмотрительных людей. "Запретить отправку данных на сервер", ну и по нажатию ваше всплывающее окно - "Эта функция недоступна. Купите полную версию."

mikemdr, 2018-06-20:
>Пароли мы не собираем, исходники проекта открыты, вы можете проверить все сами.
Это если я буду заливать прошивки, скомпилированные самостоятельно из исходников.
Вот только разработчики (renat2985) настоятельно [рекомендуют](https://github.com/tretyakovsa/Sonoff_WiFi_switch/issues/34#issuecomment-332157028)  "Не стоит самому компилировать. Используйте готовые bin файлы." А уж каких "закрытых" модулей там добавили в bin-файлы 
 к "открытым" исходникам....

renat2985, 2018-06-20:
@multiminik обдумаем.
@mikemdr, если у вас хватает опыта вы можете компилировать все самостоятельно. Но вы должны понимать что в ветке master мы ведем разработку новых версий. Которые не протестированы и могут работать некорректно, так же разные версии библиотек могут тоже влиять на работу прошивки. Поэтому более стабильные версии находятся именно в bin файлах.
Вас не заставляют пользоваться проектом. Мало того с вас даже не просят денег за наше потраченное время на разработку проекта. Если что-то не устраивает ищите другие альтернативные проекты.

tretyakovsa, 2018-06-20:
Так как код открыт вы можете видеть, что происходит при сборе данных.
`// ------------- Данные статистики -----------------------------------------------------------
void statistics() {
  String urls = "http://backup.privet.lv/visitors/?";
  urls += WiFi.macAddress().c_str();
  urls += "&";
  urls += getSetup(configsS);
  urls += "&";
  urls += ESP.getResetReason();
  urls += "&";
  urls += getSetup(spiffsDataS);
  String stat = getURL(urls);
  sendOptions("message", jsonRead(stat, "message"));
}`
Вы мне скажите. Не пугает вас то, что все это сделано на китайском железе со скрытыми функциями от производителя. И это значит, там может быть любой более страшный функционал. Вплоть до захвата управления всеми устройствами прямо по команде из центра!
Так, что с начала, стоит договорится с производителем ESP. И получив от них 100% гарантию того, что esp8266 не претендует на роль жучка глобального контроля, уже использовать нашу разработку :)

mikemdr, 2018-06-20:
> Поэтому более стабильные версии находятся именно в bin файлах.

Жаль, что выпуск новой стабильной версии переносится "со-дня-на-день" уже более полу-года.

> Вас не заставляют пользоваться проектом. Мало того с вас даже не просят денег за наше потраченное время на разработку проекта. 

За проект - БОЛЬШОЕ спасибо!!! Вот только существуют способы монетизации и "бесплатных" проектов (не совсем корректный пример: вирусы тоже - "бесплатны", а денег приносят... Надеюсь - НЕ про данный проект!)

> Если что-то не устраивает ищите другие альтернативные проекты.

Что бы понять, устраивает меня что-то - надо знать "+" и "-", а вот один из "ключевых" (для некоторых) минусов  Вы и НЕ озвучили заранее.

> Как вы не понимаете что вся эта статистика для вашего же блага. Видя эту информацию мы можем

Прям как у РКН! 

> Так, что с начала, стоит договорится с производителем ESP. И получив от них 100% гарантию того, что esp8266 не претендует на роль жучка глобального контроля,

Вот только ESP выпускаются, продаются и используются миллионами (или миллиардами?) гораздо дольше Вашего проекта - ГОРАЗДО выше вероятность того, что кто-нибудь да и заметил бы "неадекватное поведение". (При этом я на 99,9(9)% уверен что жучков там НЕТ - при таких масштабах применения АНБ и ФСБ уже давно изучили маски кристаллов под электронными микроскопами, а Сноуден и Ассанж уже всё нам рассказали бы.)

closed 2018-06-20 by renat2985