tretyakovsa/Sonoff_WiFi_switch #87: Сценарии по освещённости не работают.

multiminik, 2018-06-27:
Объясните как заставить работать сценарии по аналоговому входу.
В коде есть highalarmA0 и lowalarmA0, как понимаю это пороги сработки. В файле sensor.json прописаны пороги только для POW (highalarmpow и lowalarmpow). Добавляю для A0, но это ситуацию не меняет - сценарии не работают.
Сценарий такого плана:
Если Sonoff stateA0 < 500, тогда Sonoff relay on1.

tretyakovsa, 2018-06-27:
Обязательно нужно устанавливать две границы:
[image]
Условие это время обновления графика не стоит делать чаще 0.5 секунд.

tretyakovsa, 2018-06-27:
Сценарии имеют полный приоритет над ручным управлением. Как в интерфейсе сделать отключение в ручной режим предлагайте идеи.

multiminik, 2018-06-27:
@tretyakovsa 
Я думаю будет достаточно кнопки "Активировать сценарии \ Активировать ручной режим " в блоках над элементами ручного управления.

multiminik, 2018-06-27:
Делаю сценарии:
[image]
Верхний предел на графике отображается, а нижний так и остаётся на нуле:
[image]
Сценарий я так понял ждёт нижний предел из [data2] в analog.json( там остался 0  ),  а data2 берёт значение из lowalarmA0, но в config.options.json нет такой переменной, но есть другая:
[image]
Как быть в такой ситуации? Где-то неправильный разбор строки для формирования config.options.json?

tretyakovsa, 2018-06-29:
Дайте configs/manually.txt
и scenary.save.json

multiminik, 2018-06-29:
Вот файлы:
[manually (2).txt](https://github.com/tretyakovsa/Sonoff_WiFi_switch/files/2147815/manually.2.txt)
[scenary.save.txt](https://github.com/tretyakovsa/Sonoff_WiFi_switch/files/2147816/scenary.save.txt)

tretyakovsa, 2018-06-29:
Обновитесь по этой ссылке: http://192.168.0.XX/upgrade?spiffs=http://backup.privet.lv/esp/Sonoff_WiFi_switch.spiffs.bin&build=http://backup.privet.lv/esp/Sonoff_WiFi_switch.ino.generic.bin 
В конфигурации отключите вывод в UART.
У меня все работает. Если что стучитесь в скайп.

multiminik, 2018-07-02:
Обновился. Спасибо, всё заработало.
По поводу сценариев:
Не плохо было бы иметь возможность задавать им имена.
Выводить список в блок сценарии на главной.
Там же у каждого пункта в списке можно добавить кнопку включения\отключения.
Согласитесь, что приятнее видеть в интерфейсе имя, а не набор действий.

multiminik, 2018-07-03:
Разобрался в чём была загвоздка при сборке из исходников. Нужно было увеличить максимальную длину буфера в библиотеке "StringCommand". Значение в текстовой команде sCmd.readStr(alarmSet);
    alarmSet="ALARM "+stateA0S+" "+highalarmA0S+" "+lowalarmA0S;
попросту не помещалось в отведённые буфером 32 знака.

tretyakovsa, 2018-07-04:
Там с другими датчиками будут косяки, код еще не дописан как нужно. Библиотеку исправлю.

multiminik, 2018-07-11:
@tretyakovsa 
Скажите, а как в коде будет реализовано отключение в ручной режим? Отключение по физической кнопке рассматриваете? Есть предложение привязать к физическим кнопкам долгое нажатие и по удержанию (например в 5000мс) отключать сценарии.

tretyakovsa, 2018-07-11:
Сейчас допилим датчики в интерфейсе и будем делать переход в ручной режим. Как один из вариантов управление с кнопок принимается. Тоже будет реализовано через сценарий.