tretyakovsa/Sonoff_WiFi_switch #89: Бета версия готова для тестирования.

tretyakovsa, 2018-07-28:
Смотрите видео: https://goo.gl/mgUxAS

janisbvp, 2018-07-29:
Пашет на ура. Пока только на DH21+WS2812 проверял.
Заметил вкладку Sonoff POW - на этих устройствах тоже работает? У меня пара валяется на полке, но оригинальный софт такой ужас, что больно. 
Попробую перешить и поигратся...

tretyakovsa, 2018-07-30:
POW не совсем готов. Если будите тестировать сообщайте какие данные нужно выводить и как это отобразить на интерфейсе.

janisbvp, 2018-07-30:
Могу выслать вам на тест. Купил 2шт поигратся но так и не нашёл применение, так как датчики к нему нежелательно подключать - схема под напряжением.

gorchacov, 2018-10-15:
здраствуйте,в бета версии при подключении датчика dht22 отображается только влажность,как добавить отображение температуры?

janisbvp, 2018-10-15:
[image]

lalayan089, 2018-10-15:
Можно ли на нем сделать термостат с включением и выключением по заданным температурам?