tretyakovsa/Sonoff_WiFi_switch #93: Немного идей для улучшения

AlexSanLV, 2018-08-06:
Я случайно нашел этот проект, идея понравилась, вот несколько идей:
1. На верху страницы показывать дату и время, желательно номер сборки, можно было бы легко видеть устарела она или нет. Можно в настройках сделать сравнение прошивок на разных модулях в одной системе и рекоммендовать их обновить если выйдут новые
2. NTP - как часто обновлять? Было бы не плохо поставить период синхронизации. У меня дома свой NTP и время у всех железяк берется с него.
3. После включения и перезагрузки обязательно обновить время. Заметил разное время на модулях. Можно сравнивать его между модулями - если большая разница - обновлять время. Важно для сценариев по времени.
4. Сделать в настройках закладку "найденные устройства" - списком, с возможностью забыть какое то на случай если у него поменяется название. Я как то поменял на одном и у меня были видны оба - и старое и новое..
5. В сценариях при нажатии скажем что то включить можно сделать автоматом рефреш страницы после нажатия на кнопки - не надо перегружать вручную.
6. Думаю прикольная фишка, которую потом можно сделать платной - в сценарий добавить возможность отправку своих команд, т.е. например если реле в ванной активно, то отправляем комманду такого типа (воэможно просто линк вписываем - http....):
getURL("http://login:pass@IP/scada-remote?m=json&r=grp&fn=write&alias=0/1/1&value=true",0); например, сможем включить вентилятор в совсем другой системе.. (так у меня включается свет на контроллере HomeLynk от Schneider-Electric), а так отправляется статус в Мажордомо:    http:///command.php?qry=свет_в_спальне_над_кроватью_выключен
Уже сможем шире использовать возможности.
7. Замена фона вокруг информационного поля с серого на свой бекграунд - уже круто.. У каждого модуля свое оформление..

Что думаете?

AlexSanLV, 2018-08-06:
К пп.4 можно добавить статус OnLine/OffLine - дома иногда захожу с локалки а то и вообще с телика, и WiFi может не быть под рукой, т.е. я сразу не увижу что какой то модуль отвалился и работает в режиме AP.

renat2985, 2018-08-06:
1. Время и дата у нас была в ранний версиях. Я считаю это лишнее. NTP сегодня поправили, теперь время сравнивается при каждом запуске и 1 раз в сутки. Проблем с временем не должно быть больше. Когда появляется новая версия, у вас должен повиться такой треугольник.  Жанав на него вы установите саму последнею версию. Сегодня так же сделал несколько фиксов. В Бета версии он сейчас будет появляться всегда, нажав на него откатитесь на стабильную версию. [image]

2. Добавил более подробное описание:
[image]

6. Вы уже можете отправлять GET при включение \ выключение чего либо. Искать в сценариях.

7. Управление RGB после смены цвета фон меняется под этот цвет. Сменить цвет можете нажав на Ctrl+M и добавив такие строки: 
[image]

mikemdr, 2018-09-27:
По поводу смены цвета фона: "Ctrl+M" работает (насколько я понял) только в Chrome-подобных браузерах - а они используются не всеми. (У меня, например, дома на ноуте основной FireFox и доп.  Edge. У жены на работе админами разрешён только FF). При этом у меня на работе в  Я.браузере "Ctrl+M" на главной странице устройства НЕ отрабатывается, а на странице "Управление" - отрабатывается.

renat2985, 2018-09-28:
@mikemdr "Ctrl+M" работает во всех браузерах на страницах page.htm. На страницах index.htm (главных страницах) в старой версии эта функция еще не доступна. Но уже работает в новой бета версии.

evg26, 2018-11-09:
Добавить возможность подключать импульсные счетчики (воды, газа, и.т.д.) с возможностью корректировки показаний.

tretyakovsa, 2018-11-10:
Сделаем только для модулей с 4 Мб памяти.

evg26, 2018-11-11:
Реле. (PIN реле/Номер реле/Состояние реле после включения ESP/Игверсия) Игверсия

evg26, 2018-11-11:
Добавить возможность подключать датчики протечки воды

evg26, 2018-11-11:
Добавить возможность выводить на главную страницу контроль состояния GPIO

evg26, 2018-11-12:
Контроль состояния входов модуля ESP8266
[image]

tretyakovsa, 2018-11-12:
@evg26 Есть функция кнопок, чем не подходит?

evg26, 2018-11-12:
Как вывести состояние кнопок на главную страницу?

Jawa12, 2018-11-12:
Господа, ну помогите реализовать функцию управления импульсным краном! Это когдаодной кнопкой управляешь 2мя выходами: (к примеру вкл>импульс0 в 300мс gpio2...откл>импульс0 в 300мс gpio3). Не могу в Вашей прошивке разобраться с переменными... с лета кран так работал для полива, правда реализовал через пример IOTmanager a... Ну или куда копать-то...

evg26, 2018-11-12:
Может поможет https://github.com/letscontrolit/ESPEasy

tretyakovsa, 2018-11-13:
> Как вывести состояние кнопок на главную страницу?
@evg26 У нас нет возможности менять страницу в реальном времени.

tretyakovsa, 2018-11-13:
> Господа, ну помогите реализовать функцию управления импульсным краном! Это когдаодной кнопкой управляешь 2мя выходами: (к примеру вкл>импульс0 в 300мс gpio2...откл>импульс0 в 300мс gpio3). Не могу в [...]

Уже почти готовы импульсы. Это можно сделать будет сделать через сценарий. Может завтра закончим.

evg26, 2018-11-14:
> > Как вывести состояние кнопок на главную страницу?
> 
> @evg26 У нас нет возможности менять страницу в реальном времени.

А как сейчас отображается "Освещенность"

tretyakovsa, 2018-11-14:
> А как сейчас отображается "Освещенность"

@evg26 На графике это сделать можно. Где и в каком виде хотите смотреть?

evg26, 2018-11-22:
> Сделаем только для модулей с 4 Мб памяти.

Не получается сделать?

tretyakovsa, 2018-11-22:
@evg26 Даже не брались еще. Нужно новую версию того что есть выпустить. Пока работаем над этим.

evg26, 2018-11-23:
http://radioseti.ru/index.php?route=product/product&path=71&product_id=231

Jawa12, 2018-11-30:
Здравствуйте Господа! А не могли бы Вы добавить в функционал некоторые переменные(флаги), значения которых (0/1) можно было бы изменять через сценарии??? очень не хватает при управлении... и чтоб сохранялись в EPROM...Да ещё и в MQTT отправлялись... :) Может создать для этих целей виртуальные реле? 
объясняю как хочу использовать: при управлении импульсным реле одной кнопкой - 1е нажатие: прошёл импульс, а состояние нигде не фиксируется...пока "использую" состояние реального реле... но это как-то расточительно...
А ещё хотелось бы использовать в сценариях таймеры... Скажем: включать RGB при t>20.00 & t<23.00...
Спасибо за замечательный проект...

lalayan089, 2018-11-30:
Помоему лучше сделать в терминале набор команд для добавления/изменения любых типов переменных как в EPROM так и в RAM и в случае надобности отправкой этих переменных на MQTT с определенной частотой для каждой переменной и конечно просто просмотра логов без отправки на MQTT как это реализовано в аналогичных проектах. Хороший пример "тасмота", он на английском и функционал очень ограничен но реализация мне очень нравится особенно управление командами и отправка данных на MQTT.

renat2985, 2018-11-30:
> Здравствуйте Господа! А не могли бы Вы добавить в функционал некоторые переменные(флаги), значения которых (0/1) можно было бы изменять через сценарии??? очень не хватает при управлении... и чтоб [...]

Несколько дней назад появился нотип модуля под названием PINOUT это альтернатива RELAY но он нигде не выводится, имеет значения 1 и 0. Я думаю это то что вам нужно.

Jawa12, 2018-12-01:
@renat2985  Спасибо, пробую! 
А как насчёт использования таймеров в сценариях? Фактически хочется чтобы вкл.подсветка : 1.при движении 
2. В тёмное время( при освещённости менее...) ;
3. Только вечером(таймер)... а не ночью когда кошке вздумается побегать...:grinning:
Спасибо!

mikemdr, 2018-12-05:
Здравствуйте! В новой версии, если в Таймерах отметить чекбокс "Выполнить 1 раз", то после отработки таймера он удаляется. Может лучше добавит  чекбокс "Активен" в Список Таймеров и не удалять Таймер, а сбрасывать этот чекбокс?
Один из примеров: На даче электрообогреватель (только поддерживающий "+" в доме) каждый будний день выключается в 7:00(по окончании дешёвого тарифа). Но если мы изредка остаёмся на даче до утра Понедельника, то приходится удалять данный Таймер (чтобы не проснуться в 9:00 в холодной комнате), а потом заводить его снова, т.к. если добавить Таймер на включение в 7:00:30, мы просыпаемся от громкого "бииииип" при включении электрообогревателя. А так мы бы в Воскресение  сняли "Активен", а утром Понедельника снова его установили !

renat2985, 2018-12-06:
Сделать чекбокс в каждом таймере под названием "Активен". Да, в этом что-то есть.

renat2985, 2018-12-12:
@mikemdr сделали чекбоксы для таймеров. Уже доступно в beta (тестовой) версии. Можно обновится через страницу Debug.
[image]