tretyakovsa/Sonoff_WiFi_switch #96: Ошибка при сборке проекта

alxnovo, 2018-09-03:
Выручайте, никак теперь не могу собрать проект. Причём не собираются как старые исходники (которые компилировались) так и текущие.

Arduino IDE при компиляции выдаёт вот такую ошибку

[image]

multiminik, 2018-09-03:
Поиграйте версиями библиотеки ArduinoJson. Если обновляли на последнюю бетку, то собираться не будет. Отлично собирается на последней релизной версии библиотеки.

alxnovo, 2018-09-03:
Пробовал даунгрейдить ArduinoJson до стабильных 5-ых версий, тогда при компиляции ругается на WebServer8266, даунгрейдил библиотеки плат, то же на WebServer8266 ругается.
Потратил 3 часа ночью, но даже сборка прошлого года не компилируется, раньше она без вопросов компилировалась.
@tretyakovsa на каких версиях библиотек вы компилируете свой проект?

alxnovo, 2018-09-03:
Вот такая ошибка при старых версиях библиотек JsonArduino 5.13.12 и ESP8266 Community 2.4.1

[image]

alxnovo, 2018-09-04:
@tretyakovsa Сергей, спасибо!

Помог даунгрейд до:
       -  ArduinoJson 5.12.0
       -  главное ядро 2.3.0

Компиляция прошла успешно.

closed 2018-09-04 by alxnovo