tretyakovsa/Sonoff_WiFi_switch #100: Новый Checkbox

QLAWQ, 2018-10-28:
Предлагаю такой вот вариант чекбокса.
[image]

Его код: 
       element.innerHTML += '<div class="toggle-button-cover" '+style_val+'><div class="button-cover"><div align="center">'+renameBlock(jsonResponse, obj.title)+'</div><div class="button r '+obj.design+'"><input type="checkbox" id="'+name_val+'" '+action_val+' class="checkbox" '+checked+'><div class="knobs"></div><div class="layer"></div></div></div></div>';
      
Что скажите?

QLAWQ, 2018-10-29:
Надпись сверху. При отображении на мобильных не наползает на чекбокс.

renat2985, 2018-10-29:
Да, действительно если описание в title достаточно длинное некоторые символы попадают под кнопку.

Ваше решение к сожалению тоже не идеально, если мы будем использовать большие кнопки текст будет попадать так же под кнопки:
[image]

Добавил новый css, теперь текст будет отображаться вот так
[image]
Это не спасет от длинных предложений но текста без ущерба от дизайна теперь можно писать больше

Если есть еще интересные предложения пишите. Но старайтесь более подробно их описывать.

gorgonov, 2018-11-18:
Добавил на странице управления красивую кнопку, но она не переключает реле.Что не так?
{
"type": "checkbox",
"title": "Свет в теплице 1",
"state": "{{relay1}}",
"design":"one",
"response":"[[relay1]]",
"action":"relay not 1"
},

а на главной странице не знаю, как такую кнопку добавить.

gorgonov, 2018-11-18:
В логе пишет:
[image]

renat2985, 2018-11-18:
Вот так попробуйте:
```
 {
"type": "checkbox",
"title": "Свет в теплице 1",
"state": "{{relay1}}",
"design":"one",
"action":"/cmd?command=relay not 1"
},
```

Это скорее всего баг, я посмотрю позже, должно работать так как вы написали.

gorgonov, 2018-11-18:
Работает, переключает. Я уже такой вариант нашел. Но! Первоначальное состояние кнопки всегда On, затем при щелчке переключает. Но показывает не реальное состояние реле, а противоположное предыдущему.

gorgonov, 2018-11-18:
> а на главной странице не знаю, как такую кнопку добавить.

>

renat2985, 2018-11-18:
> Работает, переключает. Я уже такой вариант нашел. Но! Первоначальное состояние кнопки всегда On, затем при щелчке переключает. Но показывает не реальное состояние реле, а противоположное предыдущему.

Попробуйте с "state" поиграться, напишите сами туда 1 и потом 0 посмотрите какое состояние кнопка выводит. 1 это должно быть включено, 0 выключено. Если у вас наоборот, посмотрите на странице настройки PIN-ов, включите или выключите инверсию.