tretyakovsa/Sonoff_WiFi_switch #101: Таймеры

massed24, 2018-11-14:
Всем привет, здорово, проект живёт и развивается, появилось много разных плюшек, возросла стабильность. Честь и хвала разработчикам! Обновился до последней бета версии через веб, кнопкой (SPIFFS&BUILD), появилась проблема с таймерами, таймер срабатывает и возвращается после нескольких секунд, то есть, реле включилось – секунда, две – выключилось. Обновлял таким образом в конце сентября, было все нормально.

tretyakovsa, 2018-11-14:
@massed24 Сейчас проверю спасибо!

massed24, 2018-11-14:
Обновил 10 минут назад ещё раз, через веб морду, теперь вообще не срабатывает. Надеюсь поправите, успехов, спасибо.

tretyakovsa, 2018-11-14:
@massed24 Обновляйте еще раз. Должно все работать. Если что сообщайте.

tretyakovsa, 2018-11-14:
@massed24 Да и после обновления делайте рефреш веб страницы на обновленном устройстве для очистки страниц в кеш браузера. ctrl+F5 в браузере.

massed24, 2018-11-14:
Да, заработало, супер, правда не сразу, почему-то время NTP в настройках отключалось, по-новому сохранил, перезагрузил, сделал рефреш, забил таймер, и поехало как надо, будем тестировать дальше, спасибо за оперативность.

massed24, 2018-11-18:
Привет, сегодня ещё не обновлялся, а вроде в субботу, так вот, с таймерами снова проблема, они выполняются один раз, хотя в настройках выставлено всегда.

tretyakovsa, 2018-11-18:
Ошибку устранили пробуйте.

massed24, 2018-11-18:
Судя по изменениям веб морды, вы сегодня несколько раз обновления выкладывали, удобней стало можно с любой страницы зайти куда угодно, перезагрузку упростили, добавили фишки. С таймерами не то что-то, первую ESP обновил, таймеры нормально заработали, через час обновил вторую ESP таймер отработал один раз, вместо всегда, потом снова обновил первую ESP и таймер снова отработал один раз. Мне это не критично, но мы ведь тестируем, думаю, сообщу, - не помешает.

renat2985, 2018-11-18:
Спасибо за информацию.

massed24, 2018-11-20:
Браво! исправили таймеры и пульсации, ну вы блин даёте.  :)