tretyakovsa/Sonoff_WiFi_switch #102: состояние кнопок жалюзи

massed24, 2018-11-15:
Привет, в последних обновлениях на главной и странице управления, не правильно отображает состояние кнопок, именно жалюзи, помогает рефреш страницы, проверите, пожалуйста.

renat2985, 2018-11-15:
Исправили, пробуйте. Какие у вас электро жалюзи, расскажите подробней.

massed24, 2018-11-15:
Спасибо, попробую.
Использую ваш проект в жалюзи, больше года, для остановки использую концевики а не счетчик оборотов, в основном доволен, но мне не хватает записи состояния во флешь при отключении питания, знаю, тут уже обсуждалась эта тема, что это быстро убьет флешь память. Можете только для жалюзи сделать запись состояния и чтобы можно было выбрать на свой страх и риск сохранять или нет?

massed24, 2018-11-15:
Обновился, все отлично отображает.
Спасибо ребята.

massed24, 2018-11-15:
Прошивка на работающих модулях (okno),- на фото, стоит скомпилирована из исходников за 29.09.2018, стабильно работает, последнюю прошивку модуль (Test1), тестирую на макетной плате, пока без моторчиков, на диодах.

alxnovo, 2018-11-16:
> Спасибо, попробую.
> Использую ваш проект в жалюзи, больше года, для остановки использую концевики а не счетчик оборотов, в основном доволен, но мне не хватает записи состояния во флешь при отключении [...]

Поддерживаю @massed24 , не хватает записи состояния. Да, может это убьёт флеш у ESP через 10-15 лет, к этому времени будут совершенно другие технологии и модуль будет лежать в чулане на полке. Можно просто предоставить право выбора этой возможности пользователю - чекбокс с предостережением в расширенных (для продвинутых) настройках
У меня тоже рольставни на этой прошивке работают больше года. У рольставней есть концевики, но я всё-равно немного доработал код jaluisie...между переключениями поставил задержку 500ms чтобы не спалить реле (в первых тестах так и спалил одно реле и посчитал что из-за частого переключения вверх-вниз одно реле не успело разомкнуться, а другое уже сомкнулось, спасла защита двигателя). Также переменную счётчика оборотов использую как указатель через сколько секунд обесточить все реле, чтобы он не были под постоянным питанием