tretyakovsa/Sonoff_WiFi_switch #103: использование нескольких каналов для реле и дисплея для отоброжения состояния

Banzaizaizai, 2018-11-15:
Здравствуйте
видел что год назад обещали несколько каналов добавить.
Скажите, будет такая возможность?
и второй вопрос: возможно ли подключить дисплей ssd1306 для отображения состояния реле и температуры

Спасибо

Banzaizaizai, 2018-11-15:
спасибо. уже пробую)

жаль что MQTT пока нет... мир замер в ожидании)

renat2985, 2018-11-16:
В бета версии уже есть и MQTT.

Banzaizaizai, 2018-11-20:
А платные функции, которые описаны в прошивке, как поиметь не бесплатно?

Banzaizaizai, 2018-11-21:
разобрался.  из Украины как проплатить можно. пока что одну лицензию на проведение экспериментов?запасы вебманей пока что иссякли.

и параллельно еще пара вопросов: при подключении 18b20 - появляется две температуры. одна активная с показателями и второй график пустой
и второй вопрос - как температуру в mqtt передать?

с dht11 в бета версии есть ошибки в работе.  значения скачут от нормальных до заоблачный. и график в виде синусоиды получается. со стабильной версией этой проблемы нет

renat2985, 2018-11-21:
@Banzaizaizai пишите в Skype: renat2985 
18b20 - эти датчики можно подключать паролельно к одному пину, тогда вы увидите несколько графиков. Но странно что у вас 1 датчик и 2 графика. Позвоните в Skype разберемся.
В mqtt пока передавать температуру нельзя.
dht11 - такие проблемы и в стабильной версии. Мы пока не знаем как с этим бороться.