tretyakovsa/Sonoff_WiFi_switch #105: Sonoff TH16

lalayan089, 2018-11-23:
Пробовал ставить на данный девайс. Старая версия работает но как только устанавливаю sonoff loader настраиваю wifi то больше не включается. Можно ли поставить на него сразу последнюю версию и как? Спасибо.

renat2985, 2018-11-23:
Бинарники на beta версию. Которые обновляются практически вместе с исходниками на Github а иногда и раньше, можно скачать всегда по адресу:
spiffs: http://backup.privet.lv/esp/Sonoff_WiFi_switch.spiffs.bin
build: http://backup.privet.lv/esp/Sonoff_WiFi_switch.ino.generic.bin

lalayan089, 2018-11-23:
Загрузил и настроил 
spiffs: http://backup.privet.lv/esp/Sonoff_WiFi_switch.spiffs.bin
build: http://backup.privet.lv/esp/Sonoff_WiFi_switch.ino.generic.bin

Но сразу после обновления на бэту версию через дэбаг мен, он выключается и больше не включается.

renat2985, 2018-11-23:
Вы так и не сказали мне свой Flash Chip. 
Причина еще может быть в SPI Mode, если вы прошивали в DOUT. То обновляться по воздуху у вас не получится. Т.к. там лежат bin файлы для DIO.

lalayan089, 2018-11-24:
Прошивал в DOUT так как в DIO после прошивки вообще не включается. 
[image]

lalayan089, 2018-11-24:
[image]

lalayan089, 2018-11-24:
ПРОБОВАЛ И С НАСТРОЕК РАЗРАБОТЧИКА И С МЕНЮ ДЭБАГ. ПОСЛЕ ПРОШИВКИ МОДУЛЬ ВЫКЛЮЧАЕТСЯ И НЕ ПОДАЕТ ПРИЗНАКОВ ЖИЗНИ

renat2985, 2018-11-24:
Значит обновляйте только через компьютер. Возможно когда то мы будем выкладывать bin файлы и для DOUT.

lalayan089, 2018-11-24:
В  DIO прошивается но после прошивки не включается.

lalayan089, 2018-11-24:
А почему когда прошиваю в DIO он совсем не включается?

renat2985, 2018-11-24:
Звоните в skype: renat2985

renat2985, 2018-11-27:
@lalayan089 
Качаем отсюда. По этим ссылкам всегда находятся свежие тестовые версии
spiffs: http://backup.privet.lv/esp/Sonoff_WiFi_switch.spiffs.bin
build: http://backup.privet.lv/esp/Sonoff_WiFi_switch.ino.generic.bin
Сохраняем на компьютер и прошиваем через NodeMCU Flasher, в вашем случае в SPI Mode: DOUT

lalayan089, 2018-11-27:
Ок спасибо большое, буду продолжать пробовать.

renat2985, 2018-11-28:
Выяснили проблему, GPIO 14 по какой-то непонятно причине не работает. У вас я помню есть возможность повешать датчик на GPIO 4 попробуйте. Все должно заработать.
Тестируйте с DS18B20

tretyakovsa, 2018-11-29:
Обновитесь и проверьте работу датчика на 14 пине. У меня работает.

lalayan089, 2018-11-29:
Спасибо. Попробую как только будет возможность.