tretyakovsa/Sonoff_WiFi_switch #107: ошибка работы расписания

Banzaizaizai, 2018-11-23:
[image]
скрин с расписанием

[image]

подключенные реле

при наступлении времени выполнения - отрабатывает только 2-ое реле. первое нет.

renat2985, 2018-11-23:
Скорее всего причина в том что у вас на одно и тоже время несколько таймеров. Попробуйте допустим второе реле включать и выключать на несколько секунд позже или раньше.

tretyakovsa, 2018-11-24:
Да группу действий пока сделать нельзя. Это можно будет сделать через макрос. Макросы сейчас работают из сценариев. Но выполнения из таймеров еще не готово.

Banzaizaizai, 2018-11-24:
да. все правильно.  если разнести на 10 секунд - то отрабатывает.

renat2985, 2018-11-24:
Принцип работы таймеров примерно в следующем. ESP у нас не отслеживает все таймеры. Отслеживается только 1 таймер ближайший по времени к выполнению. Как только он отработает в переменную добавляется второй таймер который по очереди ближе к  выполнению. И так далее. Это позволяет делать неограниченное количество таймеров и не загружать ресурсы ESP.
Но если вы делаете 2 таймера на 1 и тоже время, второй скорее всего не отработает. Поэтому их нужно делать разными хотя бы на несколько секунд.