tretyakovsa/Sonoff_WiFi_switch #108: В течение следующей недели выйдет новаю верся. Как обновится?

renat2985, 2018-11-24:
# В течении следующей недели на главной странице вы увидите красный треугольник. Это значит появилась новая стабильная версия, и нажав на него вы ее установите.
[image]

# ОБРАТИТЕ ВНИМАНИЕ ДЛЯ УДАЧНОГО ПЕРЕХОДА
## До обновления убедитесь что у вас на странице "Конфигурации" прописаны параметры вашего WiFi. Рис. ниже.

[image]

## После обновления старые таймеры перестанут работать, вам их нужно будет удалить и сделать новые.

[image]

## Так же на странице конфигурации PIN-ов. Нельзя восстанавливать конфигурацию с облака. Вам нужно будет заново настроить все PIN-ы.

[image]

mikemdr, 2018-11-30:
Sonoff-Basic с DOUT flash'ом можно будет обновлять (не зависнут после reboot'а)?

renat2985, 2018-12-03:
@mikemdr К сожалению нет. Он зависнет.