tretyakovsa/Sonoff_WiFi_switch #109: Проблемы при работе...

Jawa12, 2018-11-25:
В режиме "Manual" включаю 3 реле, и... еср уходит в постоянный ребут...
ещё такое же поведение заметил при создании сценария с пульсацией...
вот что в терминал пишет: 

```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16    
tail 8
chksum 0x2d
csum 0x2d
v0c897c37
~ld

Test print

Exception (3):
epc1=0x4010011d epc2=0x00000000 epc3=0x00000000 excvaddr=0x400355b8 depc=0x00000000

ctx: cont 
sp: 3fff4330 end: 3fff46b0 offset: 01a0

>>>stack>>>
3fff44d0:  00000014 000005b8 000000b7 000000b7  
3fff44e0:  3fff3584 000000b7 000000b7 4010020c  
3fff44f0:  3fffaf4a 0000091b 3fff45a0 4010068c  
3fff4500:  3fffaf4a 0000091b 3fff45a0 40227a00  
3fff4510:  00000010 3fff45ac 3fff45a0 40227ab8  
3fff4520:  00000941 3fff45b8 3fff45a0 40227c21  
3fff4530:  3fffaf4a 00000000 00000000 3fff1580  
3fff4540:  00000010 3fff45ac 00000026 40228310  
3fff4550:  00000000 00000001 3fff45ac 40227a7f  
3fff4560:  3fff3584 0000051a 3fff45ac 3fff1580  
3fff4570:  3fff45b8 3fff45a0 3fff45a0 40207ed0  
3fff4580:  3fff1568 3fff45f0 3fff45ac 40208934  
3fff4590:  3fff1568 3fff45f0 3fff45fc 40208df6  
3fff45a0:  3fff5594 0000000f 00000000 3fff7344  
3fff45b0:  0000000f 00000001 3fffaf24 0000094f  
3fff45c0:  00000941 3fff85cc 0000000f 00000001  
3fff45d0:  3fff839c 0000009f 00000025 40227bdc  
3fff45e0:  3fff1568 3fff155c 3ffe9d9c 40210b4c  
3fff45f0:  00000000 00000000 00000000 3fff9b44  
3fff4600:  00000b5f 00000941 00000000 00000000  
3fff4610:  00000000 3fff8fdc 00000b5f 00000b5e  
3fff4620:  3fff78f4 0000000f 00000008 40209c1f  
3fff4630:  00000002 00000000 40207878 4020cea8  
3fff4640:  00000000 00000000 00000000 3fff3684  
3fff4650:  3fffdad0 3fff466c 3fff35a0 40212914  
3fff4660:  00000000 00000000 00000000 00000000  
3fff4670:  00000000 00000000 00000000 00000000  
3fff4680:  00000000 feefeffe feefeffe feefeffe  
3fff4690:  feefeffe 00000000 3fff367d 402287c0  
3fff46a0:  feefeffe feefeffe 3fff3690 4010070c  
<<<stack<<<
```

А ещё, при создании сценария, хотелось бы флаг какой нибудь выставлять для сохранения состояния... ну и как правильно слать на MQTT сервер состояние не лишним был бы пример...
Спасибо!!!

renat2985, 2018-11-25:
Самостоятельно компилирование или залили наши bin файлы?

tretyakovsa, 2018-11-25:
И настройки Manual покажите. Mqtt еще не закончено.

renat2985, 2018-11-26:
Давайте сделаем так, залейте bin файлы которые выкладываем вы, и посмотрите как они работают c ESP. Если проблема не исчезнет, будем разбираться.

tretyakovsa, 2018-11-27:
Вы индикацию wifi c 13 сняли? На странице настроек.