tretyakovsa/Sonoff_WiFi_switch #110: RGB PWM

Shoorup4eg, 2018-11-26:
Здравствуйте! Прежде всего спасибо вам за этот замечательный проект!
Можете описать как должна работать sRGB 5050/3528/2835 ? Я так понимаю работа ШИМ РГБ у вас реализована не на любые gpio (с аппаратной поддержкой шим, а это все за исключением gpio16), а лишь на определенные. Кнопка "включить" ("выключить") на заглавной вэб странице работает наверно только с адресными лентами (с шим лентой точно не работает). На веб странице page.htm?rgbs ползунок яркости и ползунки цветов не работают (проверяю всё осциллографом), работает только через нажатие в цветное поле этой страницы... Хотелось бы для шим ргб такой же функционал как и для адресных лент, а именно предустановленные эффекты подсветки. Да, в настройках page.htm?configs&rgb в разделе sRGB присутствует не подписанное поле со значением 100, я так полагаю что это перекочевала "скорость" от адресных лент

renat2985, 2018-11-26:
RGB ШИМ еще никто толком не тестировал.
Предполагается подключать led ленту через ULN2003 или по такой схеме:
[image]

Давайте разбираться.
Значит когда вы на странице пытаетесь выбрать нужный цвет на цветное поле он выбирается?
Выбирая там отправляется GET http://192.168.1.26/cmd?command=rgbs set aa66cc 255 - цвет в HEX, через пробел его яркость

Когда мы тянем ползунки R, G, B у нас отправляется GET цвета не в HEX формате: 
http://192.168.1.26/cmd?command=rgbs set 204 255 255 200

Сергей у нас поддерживается такой формат цвета?

По поводу странице с PIN-ами, там скорее всего ошибка, Сергей посмотри пожалуйста сколько параметров ты собираешь.
[image]

Shoorup4eg, позвоните в Skype: renat2985 быстрее выясним баги.

Shoorup4eg, 2018-11-26:
> Предполагается подключать led ленту через ULN2003 или по такой схеме...
По части электроники мне всё понятно, подключать я планирую вообще 17 метров ленты через самодельные драйверы на мощных mosfet с опторазвязкой, но сейчас для проверки мне достаточно "тыкать" щупом осциллографа в пины esp8266 и смотреть на экран осциллографа. На картинке wRGB лента, т.е. у этой ленты кроме красных, зеленых и синих есть отдельные белые светодиоды (а в прошивке у вас судя по всему обычная трёхцветная лента). И судя по картинке подключена лента на 16, 5, 4 и 0 пины, И это немного странно, т.к. 16 пин шимить не умеет, а применение нулевого (а так же 2 и 15) пина требует некоторых ухищрений. 
> Значит когда вы на странице пытаетесь выбрать нужный цвет нажимая на цветное поле он работает корректно, цвет нужный загорается?
насчёт нужного не уверен (т.к. в палитре чистого синего и зеленого цвета нет), но вроде как всё верно загорается. 
проверил по командам:
 http://192.168.0.103/cmd?command=rgbs set ff0000 255
 http://192.168.0.103/cmd?command=rgbs set 00ff00 255
 http://192.168.0.103/cmd?command=rgbs set 0000ff 255
загораются как и задумано - только один канал, но скважность сигнала неизменна при любых значениях последней цифры, т.е. яркость не меняется.
На подобные команды реакции вообще нет:
http://192.168.0.103/cmd?command=rgbs set 255 000 000 255
http://192.168.0.103/cmd?command=rgbs set 000 255 000 255
http://192.168.0.103/cmd?command=rgbs set 000 000 255 255

Ползунком яркость тоже не меняется.

Не понимаю смысла в использовании двух типов команд для по сути одних и тех же действий :)
И кнопка "включить" на странице /page.htm?rgbs не работает

Скайп уже не сегодня

renat2985, 2018-11-26:
Картинка с NodeMCU первая из более менее подходящих которую выдал Google.
На странице Конфигурации PIN-ов предлагается подключать к 2,4,5 при необходимости их можно поменять.

В Skype буду после 15:00 +2 GMT

Shoorup4eg, 2018-11-27:
Прогнал я ШИМ по всем доступным пинам (через страницу конфигурации) и к моему удивлению шим работает на всех пинах за исключением второго (на втором пине шим сигнал тоже появляется,  но модуль при этом зависает). Единственная загвоздка была с 13 пином, т.к. по умолчанию на него повешена функция индикации режимов работы вай-фая и если её отключить, то и на 13 пине шим работает.

Shoorup4eg, 2018-11-27:
Спасибо за оперативность. Исправлено всё кроме яркости.
Яркость по Вашей наводке пробовал менять запросами:
http://192.168.0.103/cmd?command=rgb set 0000ff 100 255 9
http://192.168.0.103/cmd?command=rgb set 0000ff - 255 9
http://192.168.0.103/cmd?command=rgbs set - - 255 -
http://192.168.0.103/cmd?command=rgb set 0000ff - 255
(естественно вместо 255 пробовал произвольные значения от 1 до 255) коэффициент заполнения ШИМ не изменяется и при выборе одного цвета его величина равна примерно 25% (а должна быть равной 100%). Может просто где-то масштаб нужно поменять (с 25 до 100)? :))) Потому как эти 25% это максимум что можно получить. Для наглядности осциллограмма
[image].

tretyakovsa, 2018-11-28:
Спасибо за ваше тестирование.  Коэффициент заполнения ШИМ от 0-255 даст именно 25% так как управление у ESP от 0-1023 дает весь диапазон. Яркость сейчас не влияет на результат. Нужно смешать данные от цвета и яркости так чтоб был широкий диапазон регулировки яркости. Можно конечно смешать 0-255 цветов и 1-4 уровень яркости и получить полный диапазон, Но мне кажется это нужно делать как то по другому. И у меня на этот счет нет идей. Предлагайте.

Shoorup4eg, 2018-11-28:
> Нужно смешать данные от цвета и яркости так чтоб был широкий диапазон регулировки яркости. Можно конечно смешать 0-255 цветов и 1-4 уровень яркости и получить полный диапазон...

Не уверен что правильно понял что мне предлагать. Я могу лишь по части алгоритма "подсказать" (а скорее озвучить и так очевидное):
Яркость как и интенсивность цвета определяется коэффициентом заполнения ШИМ сигнала. Выбор цвета у вас уже есть и чтобы регулировать яркость выбранного цвета нужно пропорционально для всех трёх каналов менять коэффициент заполнения. Например путем деления/умножения коэффициентов заполнения ШИМ всех трёх каналов на введенный коэффициент (который будет регулироваться ползунком на вэб странице). Чем шире диапазон значений этого коэффициента, тем шире диапазон регулировки яркости. В таком варианте диапазон цветности не зависит* от уровня яркости.

renat2985, 2018-11-28:
Опечатались здесь или GET слали такие?
[image]

Shoorup4eg, 2018-11-28:
> Опечатались здесь или GET слали такие? 

Опечатался, отправлял все запросы rgbs

> Позвоните в Skype, мы с Сергеем сейчас Online

К сожалению я сейчас не имею такой возможности - нахожусь на работе

Shoorup4eg, 2018-11-28:
> пропорционально для всех трёх каналов менять коэффициент заполнения

Это лишь теоретические рассуждения, на практике может оказаться что при меньшей яркости цвет свечения светодиодов будет менять свой спектр. 
Могу это проверить сегодня вечером с реальной лентой подключенной к модулю и регулированием питающего ее напряжением (12В от регулируемого источника питания).

tretyakovsa, 2018-11-28:
Вы просто в ручную отправляете коды от 256-1023

Shoorup4eg, 2018-11-28:
> Вы просто в ручную отправляете коды от 256-1023

попробовал - результат отрицательный. Заполнение ШИМ без изменений

Скомпилировал сегодняшний билд и прошил с файловой системой - ползунки на странице /page.htm?rgbs стали работать, но с той же верхней границей. На запросы изменения яркости по прежнему нет реакции

Shoorup4eg, 2018-11-28:
Если в файле RGBJAL.ino заменить строки 156-158 на указанные ниже, то получим полный диапазон изменения коэффициента заполнения ШИМ от 0 до 100%
```
analogWrite(getOptionsInt("rgbSPin0"), map(hexToUint32(color.substring(0, 2)), 0, 255, 0, 1023));
analogWrite(getOptionsInt("rgbSPin1"), map(hexToUint32(color.substring(2, 4)), 0, 255, 0, 1023));
analogWrite(getOptionsInt("rgbSPin2"), map(hexToUint32(color.substring(4, 6)), 0, 255, 0, 1023));
```