tretyakovsa/Sonoff_WiFi_switch #113: Что нового в новой версии 2018.12.03.zip

tretyakovsa, 2018-12-09:
https://github.com/tretyakovsa/Sonoff_WiFi_switch/releases/download/3.0/Build.Spiffs.flash.1Mb.256Kb.2018.12.03.zip
Все что было в предыдущей стабильной версии и
Множественные датчики ds18b20;
Реле может быть несколько;
Кнопок может быть несколько;
Кнопки могут работать с разными интервалами нажатия выполняя разные функции;
С работка датчиков температуры влажности освещенности по уровням;
Таймеры постоянные и одноразовые;
Формирование импульсов заданной длительности, пульсации постоянные, по времени  по количеству повторов;
Сценарии;
ИК приемник и передатчик;
RF приемник и передатчик;
Макро команды формирование сцен;
Голосовое управление;
Отправка команд от одного модуля к другому через сценарии;
Поддержка RF выключателей LIVOLO;
Поддержка RGB 5050 / 3528 / 2835 шим;
Подключение бузера и вывод тонов;
Скоро mqtt и поддержка iot-manager
Пользуйтесь :)
Работает на любых устройствах ESP8266 с DIO режимом памяти.
Если у вас ESP8285 или модули не поддерживают режим DIO при перепрошивке установите свой режим в интерфейсе ESP8266Flasher, не будет пока работать только обновление с сайта при таком варианте!

janisbvp, 2018-12-09:
Прошил. Работает. Думаю, как пометить сеноры температуры.

tretyakovsa, 2018-12-09:
Если вы об уровнях. Зайдите в сценарии и настройте уровни больше и меньше. И что делать при достижении.

janisbvp, 2018-12-09:
Извиняюсь - я не точно высказался - хотел дать сенсорам имена, чтобы знать который где воткнут. Но сдох модуль до того, как смог сделать скриншот. После перепрошивки отрабатывает полчаса и дохнет. Надо перестать экономить на ESP8266 - не хорошие эти взял - Wemos D1 mini pro.

mikemdr, 2018-12-11:
Поддержку Sonoff POW будете доделывать?

Jawa12, 2018-12-14:
Верните MQTT ! Работало же для реле...и для RGB (больше ни где не пробовал). А сейчас даже нет настроек его на странице... Спасибо!