tretyakovsa/Sonoff_WiFi_switch #115: esp-01 на p25q80h. ошибки в работе

Banzaizaizai, 2018-12-11:
Выглядит так: 
прошивается. ТД появляется,  вебинтерфейс поднимает. 
выбираем свою ТД, перезагружаем модуль,  на пару секунд плата регистрируется в сети и после этого отваливается

в эфире появляется ТД формата ESP-XXXXXX

если к ней подключится - доступен вебинтерфейс, но все пункты в нем - Undefined

скрин добавил
на хабре писали с этой микросхемой есть проблема с ФС
[image]

Banzaizaizai, 2018-12-11:
https://habr.com/post/409911/
статья с хабра

renat2985, 2018-12-11:
Да, вам не повезло с Flash Chip. Как заставить правильно писать в файловую систему PUYA чипы, я не понимаю.
Делал подобные эксперименты https://github.com/esp8266/Arduino/issues/4061#issuecomment-422849071 не помогает.

Если есть какие-то другие идеи пишите.
У меня тоже много ESP PUYA.

renat2985, 2018-12-11:
@Shoorup4eg конкретно с этим проектом (Sonoff_WiFi_switch
) почему-то не работает.

renat2985, 2018-12-14:
@Banzaizaizai, попробуйте эти bin файлы на ESP с PUYA Flash чипом.
Только прошивайте в режиме DOUT. И отпишитесь как работают.

build http://backup.privet.lv/esp/Sonoff_WiFi_switch.ino.generic.puya.bin
spiffs http://backup.privet.lv/esp/Sonoff_WiFi_switch.spiffs.bin

Banzaizaizai, 2018-12-15:
в понедельник возьму из магазина, прошью и отпишусь что да как.. самому интересно. пока что поменялся на человеческую версию

Banzaizaizai, 2018-12-20:
попробовал. не поменялась ситуация. регистрация в сети на пару секунд и досвидос.  поднялась ТД ESPxxxx

renat2985, 2018-12-20:
@Banzaizaizai у меня аналогично. Ладно, может что-то еще в голову со временем придет.

antkachenko, 2018-12-26:
У меня тоже таких модулей много пришло. Решаю проблему покупкой модулей памяти Winbond в Chip и Deep и перепайкой...