tretyakovsa/Sonoff_WiFi_switch #117: Сценарии - какие опции доступны в бесплатной версии?

antkachenko, 2018-12-13:
При нажатии на кнопку 1 (см. скриншот) выдает сообщение 2. Правильно ли я понимаю, что сценарии доступны только в платной версии?

Скриншот: http://joxi.ru/xAen6WyCpN5WOr

renat2985, 2018-12-13:
Платная версия отличается только возможность редактировать страницы я о Ctrl+M когда откроется редактор страницы, где вы можете изменить интерфейс, вы сможете изменить там код, сможете просмотреть страницу с измененным кодом, но не сможете сохранить этот код.
И в бесплатной версии в разделе "Создание сценариев" недоступна кнопка с лупой которая в некоторых случаях позволяет проще создавать сценарии, нажав на нее она позволяет узнать текущее состояние нужного элемента (relay, rgb, ir и т.д.) на работоспособность сценариев это никак не влияет, просто сложнее иногда создать сценарий.
Вы можете сами вписать в окошко нужный вам параметр.

antkachenko, 2018-12-13:
А как узнать возможные варианты триггера "Состояние WiFi"?

renat2985, 2018-12-13:
В платной версии вы тоже не узнаете возможные варианты.
Предлогайте где их и как выводить?
"Состояние WiFi" вот они:
```
    // 0 WIFI_EVENT_STAMODE_CONNECTED    подключение к роутеру получение ip
    // 1 WIFI_EVENT_STAMODE_DISCONNECTED попытка переподключения к роутеру 
    // 2 WIFI_EVENT_STAMODE_AUTHMODE_CHANGE
    // 3 WIFI_EVENT_STAMODE_GOT_IP подключен к роутеру
    // 4 WIFI_EVENT_STAMODE_DHCP_TIMEOUT Не получен адрес DHCP
    // 5 WIFI_EVENT_SOFTAPMODE_STACONNECTED подключен клент
    // 6 WIFI_EVENT_SOFTAPMODE_STADISCONNECTED отключен клент
    // 7 WIFI_EVENT_SOFTAPMODE_PROBEREQRECVED Режим точки доступа
    // 8 WIFI_EVENT_MAX,
    // 9 WIFI_EVENT_ANY = WIFI_EVENT_MAX,
    // 10 WIFI_EVENT_MODE_CHANGE
```