tretyakovsa/Sonoff_WiFi_switch #118: Графики за длительный период времени

BotovNikita, 2018-12-14:
Отличный проект!
Есть ли возможность отображать график температуры за выбранный промежуток времени? 
К примеру модуль установлен в бойлерной, и управляет насосами по температуре. Было бы здорово иметь возможность просмотреть историю изменения температуры, знать когда какой сценарий сработал, знать состояние реле в тот или иной момент времени.

renat2985, 2018-12-14:
Спасибо. Мы начали это делать, но как скоро закончим не известно.
Но пока вы можете как и я сохранять нужнфе вам параметры через ifttt.com например в google spreadsheet (таблицы)

Регестрируемся на https://ifttt.com далее:
[image]
Вам нужно активизировать Webhooks сервис и вы получите секретный ключь для URL.
Далее создаем сценарий:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Далее открываем новую вкладку в браузере и октрываем на вашем ESP устройсве эту ссылку:
http://192.168.1.*/config.live.json
Здесь вы увидите все живые данные, те данные которые постоянно меняются, тимпературу, состоние кнопок и т.д.
[image]
Запоминаем имена нужных вам сенсоров, например temperature1 и A0
Открываем еще 1 вкладвку в браузере и открываем сценарии:

Создаем нужным вам сценарий, вы можете сделать например сценарии на включение реле, которое включает боллер, далее в "Тогда" выбираем "GET" и ниже формируем ссылку на ifttt.com
У меня она получилось такой:
http://maker.ifttt.com/trigger/sensor/with/key/СЕКРЕТНЫЙ_КЛЮЧЬ/?value1={{A0}}&value2={{temperature1}}
Обратите внимание у нас http:// а не https://

[image]

На этом все. После открываем https://drive.google.com/
Ищем файл с названием ESP-spreadsheet
В нем будет примерно следующее:
[image]

BotovNikita, 2018-12-14:
Большое Спасибо за подробную инструкцию!
Успехов в реализации всех фич!)

Aval0ncheG, 2020-01-11:
@renat2985 подскажите, как проще всего сделать что бы GET запросы постоянно отправлялись? К примеру раз в 10 или 30 минут, 50 таймеров не охота создавать)

Aval0ncheG, 2020-01-16:
Свою проблему я решил следующим образом.
Создал два сценария
[image]

renat2985, 2020-01-16:
Свяжитесь со мной в skype: renat2985