tretyakovsa/Sonoff_WiFi_switch #121: Обновление двух старых SonOff

i81, 2018-12-30:
Здравствуйте!
Порядка года тому мне всем миром помогали воскресить два реле, которые умирали по непонятной причине https://github.com/tretyakovsa/Sonoff_WiFi_switch/issues/65 
Нашёл я их недавно - включил - работают оба, но возникла проблема: необходимо мне изменить имя точки доступа и тут засада: 
1. Из web интерфейса настройки реле сеть не меняется. Хотя фактически сохраняется и остаётся новая, но даже после перезагрузки подключается к старой и, если нет старой - создаёт свою открытую.
2. Проблема, описанная мной тогда, в старом топике, осталась - к открытой точке подключится не получается ни Win10, ни Андроид не может.
3. Одно реле я попробовал обновить через восклицательный знак (только потом сюда зашёл и прочитал, что этого делать не стоит) после попытки обновления реле включается, но web странице пишет FileNotFound и всё :(
Собственно вопрос у меня как, не имея программатора, попытаться воскресить реле после обновления. Я помню Ренат мне когда-то подсказывал длинную ссылку по которой можно было в реле закачать прошивку, но вот саму ссылку сейчас не нейду :(

i81, 2018-12-30:
[image]
Судя по всему у меня на том, втором, рабочем, который я не покалечил обновлением, прошивка 2017.10.16 но почему-то она не подсвечена жирным, о чем написано в соседнем топике.
Может кто подскажет как поменять сеть на живом устройстве?

solalex1, 2019-01-22:
Таже самая беда на старом sonoff-e после перехода на последнюю прошивку не меняется wifi сеть, хоть ты убейся. C разных браузеров прописывал новую точку - настройки нового подключения не сохраняются ни в какую. Может есть команда сброса на дефолт? Чтобы не было прописано ни одной точки доступа.

Egor2101, 2019-01-22:
Я перед заливкой других прошивок использую flash_download_tools_v3.6.5 Кнопочка erase. Чистит все...