tretyakovsa/Sonoff_WiFi_switch #122: Голосовое управление как работает?

Jawa12, 2019-01-02:
С новым годом, господа!  В меню прошивки заложена возможность голосового управления... Но нигде не нашел описания, как осуществить задуманное? Какое оборудование,ПО используется при этом? Да и вообще, работает ли? Спасибо...

tutaf, 2019-01-02:
@Jawa12 с Новым Годом! Эта функция позволяет esp принимать команды вида http://<your_ip>/voice?command=<your_command> и выполнять полученную команду, если она приписана в сценарии.
Сейчас пилю приложение под android, которое при определённом событии(например, нажатие кнопки в уведомлениях или произнесение фразы голосовой активации (с последним пока есть проблемы)) запускает распознавание речи и отсылает результат на прописаную в настройках esp.

tretyakovsa, 2019-01-02:
Нужна программа которая голосовую команду может отправить как Get запрос по адресу http://<your_ip>/voice?command=<your_command>.

renat2985, 2019-01-13:
Видео о том как работает голосовое управление: https://www.youtube.com/watch?v=hTob3RDLrNc