tretyakovsa/Sonoff_WiFi_switch #124: Как Настроить несколько Реле на одной ESP ?

ds32, 2019-01-02:
Не нашел видео\гайдов по настройке реле.

вопрос №2 - Можно ли сделать цикличный таймер обратного отсчета? что бы  менялось состояние кнопки с регулируемой выдержкой, к примеру 10 min ON, 2 min OFF   (аналог timer555, только программный) 

Вопрос №3 - Интересно стало, а что будет если нажать Udpate, нажал, после авторебута не поднялось, даже спустя 10 мин. Часть настроек сохранилось:
"Подключение к Wi-Fi роутеру"  Пункты SSID и Password пусты но к роутеру законнектилось - Баг или Фича ? :)  Есть ли возможность затереть все настройки через веб - что то типа "Reset Configuration", что бы не затирать всю флешку каждый раз после экспериментов?

p.s.
С Новым Годом!!! :)
01.01.19:
2 часа ночи: попался на глаза гайд по смене пина реле, а я как раз задумывался собрать на ESP-12F.
3 часа ночи: собрал на макетке тестовый вариант, встроенный прошивальщик с настройками по умолчанию - wifi-ap не поднял, по этому запустил привычную ESP_DOWNLOAD_TOOL_V2.4 (именно 2.4) (автонастройка там удобная, сама выставила: 26MHz, QIO, 32mbit)
4 часа ночи, настроил одно реле - работает!!!  :+1:, осталось еще как то 3 реле подцепить.

artgareev, 2020-06-03:
Возможно задействовать большее количество реле? 
Например на Esp8266 Lolin 
[image]

tretyakovsa, 2020-06-03:
@artgareev  Если ног хватает то да. Gpio  0  2 4 5 11 12 13 14 15 16 Все будут работать с учетом того что вы реле правильно подключите.  например с 15 0 и 2 GPIO в лоб подключение может и не получится если, вы не понимаете в схемотехнике.
GPIO 6 7 8 9 10 заняты flash и туда ничего нельзя подключать.

artgareev, 2020-06-03:
А как настроить в интерфейсе чтобы было более 4х?
[image]

tretyakovsa, 2020-06-03:
@artgareev  Отредактировать с помощью редактора /edit файл с конфигурацией потом сохранить в редакторе ctrl+s