tretyakovsa/Sonoff_WiFi_switch #125: Нужны мануал и описание!

multysea, 2019-01-03:
Подскажите, есть ли  где-либо доступная для чтения или скачивания инструкция по эксплуатации этой системы? Подробно описывающая функционал и ограничения, со схемами подключения и т.п. Здесь вроде бы все перерыл, не нашел. Спасибо!

tretyakovsa, 2019-01-03:
Пока отдельного нет. Все разрознено. Установить получилось?

renat2985, 2019-01-03:
Некоторая информация есть в wiki: https://github.com/tretyakovsa/Sonoff_WiFi_switch/wiki 

Так же можете попробовать посмотреть виде:
https://youtube.com/playlist?list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi

Вопросы связанные с подключением можете задавать здесь в новых issues.

renat2985, 2019-01-03:
Скачать bin файлы можно с главной страницы нашего репозитория. Или вот прямая ссылка  https://github.com/tretyakovsa/Sonoff_WiFi_switch/releases/download/3.0/Build.Spiffs.flash.1Mb.256Kb.2018.12.03.zip 

В архиве так же есть приложение для заливки:
[image]

Из исходников собирать не стоит.

tretyakovsa, 2019-01-03:
Вы установите и если что будет не понятно пишите сюда.

multysea, 2019-01-11:
Залил прошивку в плату WeMos D1 mini  (8266ex), все корректно отработало. А как подключить теперь плату к моей точке доступа:-)? Обычно делал это с помощью ESP8266 SmartConfig, но на этой плате кнопки flash нет, есть только кнопка reset..

multysea, 2019-01-11:
Залил прошивку в плату WeMos D1 mini, все корректно, зашел, настроил wi-fi и ddns, сервер отзывается, все как бы прекрасно. Но, не смог ни датчика подключить, ни реле. Датчик назначил на d1, реле (как бы, пока просто светодиод) - на d8, все сохранилось, перезагрузил плату, но ни температуры не видно, ни диод от кнопки не загорается.  Прошу помощи:-).

multysea, 2019-01-11:
[image]
[image]
[image]

tretyakovsa, 2019-01-11:
@multysea Есть много настроенных конфигураций и если нет крайней необходимости лучше самому пины не настраивать используйте предустановленные (пока не разберетесь в тонкостях платы).
Ваша ошибка в том, что мы считаем пины как GPIO, а вы пытаетесь воспользоваться именами D.
Тому кто это придумал ручки бы оторвать.
Смотрите соответствие:
[image]

renat2985, 2019-01-11:
[image]

multysea, 2019-01-12:
Спасибо, исчерпывающе, в таком варианте - работает). Именно поэтому было бы здорово, если бы у разработчиков была возможность и находилось иногда время для составления описания и руководства по настройке и эксплуатации. Потому, что вопросов - куча, и уверен, не только у меня. Чем дальше в лес - тем толще партизаны:-).. Но раз пока мануала нет - буду спрашивать здесь).

multysea, 2019-01-12:
После перезагрузки роутера плата к нему сама не подключается, приходится ее перезагружать, тогда все ок. Что-то можно сделать?
Это один вопрос, а второй - а как воспользоваться какой-то из стандартных конфигураций? Где их скачать?
И еще, есть ли возможность использовать телеграм-бота для управления платой? если да - то как? Спасибо!

multysea, 2019-01-12:
Подключил второй ds18b20, заведомо исправный, но он нигде не появился. Первый есть на главной странице. Как вывести показания с других датчиков?

renat2985, 2019-01-12:
На каждый вопрос создавайте новую тему.

tretyakovsa, 2019-01-12:
@multysea 
> После перезагрузки роутера плата к нему сама не подключается, приходится ее перезагружать, тогда все ок. Что-то можно сделать?
Это свойства схемотехники платы. Попробуйте подключить устройство к обычному заряднику не к компьютеру. Может поможет
> Это один вопрос, а второй - а как воспользоваться какой-то из стандартных конфигураций? Где их скачать?
Скачивать не нужно просто попробуйте поменять тип устройства и смотрите что куда подключать.
[image]
> И еще, есть ли возможность использовать телеграм-бота для управления платой? если да - то как? Спасибо!
Нет и не будем делать. 
> Подключил второй ds18b20, заведомо исправный, но он нигде не появился. Первый есть на главной странице. Как вывести показания с других датчиков?
Подключите все датчики параллельно на один пин.