tretyakovsa/Sonoff_WiFi_switch #126: RF приемник 433MHz

Forldol, 2019-01-07:
Здравствуйте, прикрутил RF приемник, и столкнулся с такой проблемой - если включить реле с пульта через приемник то выключить его через веб морду уже не получиться, сценарий не дает. Я так понял это из-за того что в rfReceived значение сохраняться навсегда. Можно ли сделать что бы это значение  через определенное количество мили секунд менялось на ноль? Или это не поможет? Можно ли что- то с этим сделать?

tretyakovsa, 2019-01-09:
@Forldol Думаю да. Как поправлю напишу.

Forldol, 2019-01-10:
@tretyakovsa, спасибо, буду ждать.

Forldol, 2019-01-13:
Еще обнаружил такую неприятную особенность при работе с  RF приемником. 
Сделал сценарий такого рода: 
Если Беспроводной RF приемник = 7856476 
Тогда Управление реле not 1

при кратком нажатии на кнопку пульта ничего не происходит, если подержать дольше то релешка быстро включается и выключается. То есть нужно ловить момент  когда отпустить кнопку что бы реле осталось включенным или выключенным. С этим можно что-то сделать?

renat2985, 2019-01-14:
@Forldol, я эти проблемы решил используя такой программируемый RF приемник:
https://ru.aliexpress.com/item/433mhz-Wireless-4-Channel-RF-Receiver-1527-Learning-Code-Decoder-Module-For-Remote-Control/32828971724.html
Он поддерживает до 4 каналов. Записываем в него нужные вам коды кнопок и цепляете его ножки к ESP и в прошивке эти Pin-ы ставим в Tach (кнопки). С кнопками таких проблем нет.
Конечно костыль. Но он отлично работает.

tretyakovsa, 2019-01-16:
@Forldol Я как сделаю исправление все будет работать. А те что Ренат предлагает использовать ограничивают возможности и их применять если они уже есть на руках и девать их некуда.