tretyakovsa/Sonoff_WiFi_switch #127: Как отобразить температуры с нескольких датчиков?

multysea, 2019-01-12:
Подключил несколько DS18b20, параллельно, все на один пин, подтягивающий резистор один на все датчики. Но на главной странице отображается температура только с одного датчика. Подозреваю, что где-то должна быть конфигурация страницы, и что где-то можно настроить и названия для датчиков, и сколько каких отображать, но не нахожу.

multysea, 2019-01-12:
[image]
[image]
[image]

Файл modules.json:
 {"module":["upgrade","relay1","relay2","relay3","ntp","ddns","timers","temperature1","temperature2","A0","tach0","irReceived","irTransmitter","admin","pulse"],"SSDP":"","space":"","ip":"192.168.1.148","lang":"ru"}

multysea, 2019-01-12:
Замечу, что именно этот вариант подключения использовался сегодня в другом проекте, и в нем датчики видны.  Если нечетко видно, то там один датчик просто в плате, а второй - через клеммник, параллельно первому, вынесен на проводе в капсуле..

multysea, 2019-01-12:
Четкий визит эффект! Ни с того, ни с сего второй датчик стал показываться. Единственное, что я сделал - это перезагрузил плату в очередной раз на всякий случай.

renat2985, 2019-01-12:
Скорее всего у вас был просто плохой контакт.

multysea, 2019-01-12:
Спасибо! Не уверен в этом, хотя и не исключаю, я действительно шевелил контакты при фотографировании. Подскажите, а как назвать датчики температуры, чтобы на главной странице было видно, какой датчик где находится? А также,  в окошке датчика рисуется график во времени, а как посмотреть график за длительное время, это возможно?

multysea, 2019-01-12:
И подскажите, пожалуйста, что означают две цифры - 0 0  справа от показаний температуры?
[image]

multysea, 2019-01-12:
Благодарю! А скажите, не планируете сделать возможным индивидуальное именование датчиков?

renat2985, 2019-01-12:
Когда ни будь сделаем. Не совсем понятно куда вписывать имена датчиков тимпературы.
На странице конфигурации пинов, мы уже начали внедрять возможность дать имя Relay, PinOut, Tact, но пока это в интерфейсе не показывается. Т.к. тоже не совсем понятно куда именно эту надпись выводить, вместо надписей "Вклчить\Выключить" но тогда не будет понятно у нас выключено или включено, или перед кнопкой.
Возвращаясь к DS18B20, на странице пинов мы там просто ставим галочку на 1 элементе. И куда писать имена датчиков?

multysea, 2019-01-12:
Но все датчики 18b20 включены параллельно, при этом у каждого свой уникальный адрес. На странице пинов ничего писать не нужно, а вот на главной странице, когда датчики уже в системе видны, слово "температура" было бы хорошо иметь возможность отредактировать в соответствии с потребностями. А то когда 10 датчиков, соответственно, 10 полей и все "температура", то не очень понятно, температура чего именно индицируется..

renat2985, 2019-01-12:
Я вас понимаю. Вы меня не поняли. Перечитывайте...