tretyakovsa/Sonoff_WiFi_switch #129: Поле "освещенностть и датчике

multysea, 2019-01-12:


multiminik, 2019-01-12:
Подключите пин назначенный на "Освещённось" к земле. В качестве датчика неплохо смотрится обычный фоторезистор. Как подключить - куча инфы в сети.

renat2985, 2019-01-12:
Отключить \ Выключить можно на странице конфигурации пинов:
[image]

Подлючать можно по такой схеме:
[image]
Но правельней как советует @multiminik через резистор. Я подключаю через 1kOm. Но все это зависит от самого фото резистора, нужно подбирать.

multysea, 2019-01-12:
А0 уже нашел, но спасибо, что уделяете столько внимания!