tretyakovsa/Sonoff_WiFi_switch #131: Приложение IoT

multysea, 2019-01-13:
В вашем сегодняшнем видео сказано, что после установки приложения появится две иконки, одна браузер и вторая для управления. Но после установки с маркета у меня появилась только иконка браузера (IoT Finder).
А также вопрос, можно ли сделать как-то, чтобы приложение работало с платой не только из внутренней сети, но и снаружи?

tutaf, 2019-01-13:
Появилась только одна иконка? Она так и называется - iot room finder? Проверьте версию приложения, она должна быть 2.0. В маркете уже сутки как должна быть свежая версия.

По идее можно управлять и не находясь в сети, только укажите адрес, куда запросы надо слать. Не знаю если будет работать, проверить не могу.

renat2985, 2019-01-13:
Можно у себя в локальной сети поставить VPN сервер. Далее в телефоне настраиваем VPN на домашний сервер. И вуаля у вас безопасный туннель и возможность управлять вашими ESP как с браузера так и через это приложение. :)

renat2985, 2019-01-14:
Андрей, я понял в чем проблема у @multysea.
Сейчас устанавливал приложение на планшете. После установки на рабочем столе действительно появился ярлык "IoT Room", о котором и говорит @multysea. А если зайти в "Приложения" (там где все программы/игры) я там вижу 2 программы "Управление" и "Поиск устройства."
Вопрос, Мы можем создать 2 ярлыка на рабочем столе "Управление" и "Поиск устройства"?
Если нет. То нужно "Поиск устройства" переименовать в "IoT Room" или наоборот.
И предлагаю иконку приложения та что в Google Play поставить и в "Поиск устройства", т.к. это тоже путает, установил одно. А получил другое.

tutaf, 2019-01-14:
@renat2985  так это вроде нормально. Если в списке всех приложений есть иконка "Управление", то в большинстве лаунчеров её можно перетащить на рабочий стол. @renat2985 @multysea какие лаунчеры вы используете?

Иконку можно поменять, не проблема.

renat2985, 2019-01-14:
С "Управлением" все хорошо. Его трогать не надо.
На планшете Android 4.4.2, Launcher стандартный от Android.
Когда я установил приложение я вижу иконку IoT Room, смотри скриншот ниже. 
[image]

И когда я захожу в "Меню" (Приложения) там где показываются все установленные программы, я там вижу 2 программы "Поиск устройств" и "Управление".
В чем разница между программой "IoT Room" с иконкой как в Google Play и  программой "Поиск устройств" иконка в виде лупы? Насколько я понимаю ее нет. Значит и называть IoT Room нужно поменять на "Поиск устройств", и иконку тоже, иначе это путает.

tutaf, 2019-01-14:
@renat2985 иконка с текстом - иконка приложения. Иконка с лупой - иконка экрана(активности) поиска esp. С микрофоном - активности голосового управления. Ваш лаунчер показывает иконки приложений на рабочем столе, а не активностей.

multysea, 2019-01-14:
Снес, скачал заново, установил.  Появились: на главном экране IoT , в приложениях - Управление и Поиск устройств. Немного необычно, но в общем-то все работает).

renat2985, 2019-01-14:
@multysea что вам кажется необычным?

multysea, 2019-01-14:
Необычным является то, что приложение, которое называется IoT появляется ТОЛЬКО на главном экране, и не появляется в папке программ. А также то, что приложения Управление и Поиск устройств, наоборот, не появляются на главном экране, хотя при установке программ с маркета на главном экране должны появляться все установленные программы.

renat2985, 2019-01-14:
@multysea я с вами полностью согласен.
Андрей, давайте "Иконку с лупой - иконка экрана(активности)" Переименуем в Iot Room и дадим ей иконку с текстом. Иначе люди путаются.

tutaf, 2019-01-14:
Залил обновление с исправлениями в Google Play

renat2985, 2019-01-15:
Спасибо. Ждём пока Google одобрит. Пока ещё там старая версия.

tutaf, 2019-01-15:
@renat2985 ещё вчера вечером одобрил...