tretyakovsa/Sonoff_WiFi_switch #133: Работа через ddns

multysea, 2019-01-15:
Обнаружил, что при заходе на espчерез ddns (freemyip.com) сервер отдает не всю страницу, начинает грузить бодро, но останавливается. Скрины приложу, с мобильного не удается.

renat2985, 2019-01-15:
Все правильно. Почитайте это, и посмотрите видео ролики указанные в том issues https://github.com/tretyakovsa/Sonoff_WiFi_switch/issues/57

renat2985, 2019-01-15:
На видео у меня ушло 6-10 минут. На мануал с всеми скриншотами уйдет значительно дольше.
Видео можно сделать на весь экрна и вам будет все видно.

renat2985, 2019-01-15:
Вы задаете очень хорошие вопросы, многие из них я помечаю как wiki (часто задаваемые вопросы) эти ярлыки кликабельны.
[image]

И в Wiki появилась новая ссылка "Часто задаваемы вопросы". Большинство из них ваши. 
[image]