tretyakovsa/Sonoff_WiFi_switch #135: Датчик температуры ds18b20 и сценарии

Forldol, 2019-01-16:
Создаю сценарий: при достижении температуры включать реле но ничего не происходит. Температуру датчик показывает. 
И еще одна проблема. Через какое-то время работы значение температуры становиться -127, помогает только перезагрузка устройства. 
Датчик подключал по схеме gnd vcc, data на gpio2, между data и vcc резистор на 4,7кОм.
Устройство sanoff basic. На сколько я понял с памятью PUYA потому что прошилось нормально только с DOUT.

renat2985, 2019-01-16:
После создания сценария с Температурой, Влажностью или Аналоговым выходом. Нужно перезагружать ESP, - этот баг постараемся в течение месяца исправить.
По поводу -127 это странно.
Была бы у вас PUYA, в 90% вы бы не смогли даже к роутеру подключится. В новых моделях sonoff просто используется немного другой чип flash памяти, который хорошо работает только в DOUT.

Forldol, 2019-01-16:
> @renat2985
> Создал сценарий:
> Если temperature1 >= 25
> Тогда Пульсация on Состояние реле 1 400 400 10i
> 
> Сценарий работает если поднять температуру выше 25 и перезагрузить устройство, то есть температура в момент включения должна быть >= 25. Отрабатывает пока не упадет ниже 25 и больше не работает, если еще раз поднять выше 25 - ничего не происходит.
> Поставил десять импульсов, но по истечении 10-ти импульсов релешка всё равно продолжает щелкать пока температура не упадет ниже 25. Как то так :(
> 
> На счет температуры -127, скорее всего был датчик бракованный, поменял датчик, пока таких глюков не было.

Еще раз перезагрузил устройство, всё работает :)

tretyakovsa, 2019-01-17:
Выставляйте два уровня. Больше и меньше. Никаких больше или равно и меньше или равно. Обязательно два уровня. больше меньше. То что это случайно работает с больше или равно это моя ошибка не предусмотрел. Сценарии будут работать только после перезагрузки модуля. Но это пока, потом сделаю как надо. Может и по другому уровни будут задаваться. Если у вас работают варианты без второго уровня и с равно это баг.

renat2985, 2019-01-17:
Может в таком случае убрать <= и >= т.к. это не работает и только путает пользователей?

Forldol, 2019-01-17:
@tretyakovsa, странно, у меня работало с >=, правда через раз как-то. Сделал как Вы сказали. Всё работает как часы. Спасибо.