tretyakovsa/Sonoff_WiFi_switch #138: Поддержка модуля DS3231

Forldol, 2019-01-17:
Планируется ли сделать поддержку DS3231?
Возможный вариант, если недоступна синхронизация по NTP, то пытаемся читать время из локальных часов RTC. 
Была ситуация когда ночью отключили свет и включили его, провайдерское оборудование еще какое то время не работает, интернета нет несколько часов, а то и больше, следовательно таймеры не срабатывают.

augusev, 2019-01-20:
Поддержка DS3231 вещь очень нужная. А для серьезных проектов - необходимая!

augusev, 2019-01-20:
> Напишите почему это так важно. Хоть бы один пример серьезного проекта.
> Поднимите NTP на роутере и для серьезных проектов роутер на бесперебойник.
> Туда же USB модем с резервным каналом в интернет и флешку для сбора данных.

1. Почему так важно. Представьте котельную в загородном доме, где вы не
живете постоянно. Настроено расписание на неделю. Тракторист порвал кабели.
Пропал свет, кончился интернет. Потом свет появился, а интернет нет.
Расписание котельной как будет работать?
Настроен таймер вечернего освещения имитации присутствия. Свет пропал,
потом появился. Интернета нет. И что?
2. "Поднимите NTP....". Можно еще сделать три независимых электроввода с
резервированием дизельгенератором. И провести интернет от трех разных
провайдеров. Часового в будку поставить.
Я только сказал, что поддержка модуля DS3231 в серьезных проектах
необходима. При этом стоит этот модуль 60 рублей с доставкой. Батарейка от
30 рублей. Я же не предъявляю Вам претензии. У Вас хороший проект и многим,
как и мне, нравится. Если прикрутите DS3231 - хорошо! Ну а нет - значит
нет. Я, как и многие здесь, не программист (далеко не программист). И сам
прикрутить модуль DS3231 не смогу.

augusev, 2019-01-20:
1. Почему так важно. Представьте котельную в загородном доме, где вы не живете постоянно. Настроено расписание на неделю. Тракторист порвал кабели. Пропал свет, кончился интернет. Потом свет появился, а интернет нет. Расписание котельной как будет работать? 
Настроен таймер вечернего освещения имитации присутствия. Свет пропал, потом появился. Интернета нет. И что?
2. "Поднимите NTP....". Можно еще сделать три независимых электроввода с резервированием дизельгенератором. И провести интернет от трех разных провайдеров. Часового в будку поставить.
Я только сказал, что поддержка модуля DS3231 в серьезных проектах необходима. При этом стоит этот модуль 60 рублей с доставкой. Батарейка от 30 рублей. Я же не предъявляю Вам претензии. У Вас хороший проект и многим, как и мне, нравится. Если прикрутите DS3231 - хорошо! Ну а нет - значит нет. Я, как и многие здесь, не программист (далеко не программист). И сам прикрутить модуль DS3231 не смогу.

kostyamat, 2019-05-28:
Поддержу автора топика.