tretyakovsa/Sonoff_WiFi_switch #139: Отключение таймаута на подключение к WiFi

antkachenko, 2019-01-17:
Коллеги,

Имею следующую ситуацию: реле с управлением от ESP-01 управляет замком на входной двери. При текущих возможностях прошивки, если создать помехи на частоте 2.4Ghz на протяжении максимум 4 минут модуль перейдет в режим повторного подключения к WiFi и начнет вещать свою сеть. Это дат возможность осведомленным злоумышленникам подключить модуль к своей сети и открыть несанкционированно дверь.

Хотелось бы, чтобы можно было отключить переход на вещание собственной сети, даже при условии, что для смены SSID придется перепрошивать модуль.

renat2985, 2019-01-17:
На странице конфигурации вы можете указать пароль и изменить SSID AP на нужный вам.
[image]

@AndrewMyName можете попробовать нажать Ctrl+M и изменить 240 например на 3600, в таком случае ESP будет пытаться подключится в течение часа. Насколько я помню ранее были с этим какие-то проблемы. Возможно сейчас их нет. В таком случае мы добавить в это выпадающее меню новый пункт, подключатся к роутеру в течение 1 часа или двух.
[image]

multysea, 2019-01-19:
Ну а если вообще интернет пропал на сутки, или больше, то как заставить систему восстановить подключение без участия человека?

renat2985, 2019-01-19:
@multysea, вы немного путаете, ESP не нужен Интернет. Ей нужен доступ к вашему роутеру. Если она не видит ваш роутер через энное количество времени которое вы указали на странице, (скриншот выше) она перейдет в AP. И вам никто не мешает изменить те данных на более большие, на день, два или неделю.