tretyakovsa/Sonoff_WiFi_switch #141: DHT21

augusev, 2019-01-19:
Ренат, подскажите пожалуйста, при подключении  DHT21 в веб интерфейс выводятся значения : "Температура 28, 0, 0"  и  " Влажность 44, 0, 0". Первые две цифры это температура и влажность. А остальные цифры идущие через запятые это что?﻿

renat2985, 2019-01-19:
Это минимальные и максимальные значения в сценариях.
Когда вы создадите первый сценарий связанный с температурой, влажностью или аналоговым выходом. Например такой:
[image]

То график будет уже такого вида, где вместо последнего нолика мы уже видим те 20 которые у казали в сценарии выше. Так же здесь появилось 2 линии на графики, одна из них эта опять же те 20.
[image]

**ПОСЛЕ СОЗДАНИЕ СЦЕНАРИЯ СВЯЗАННОГО С ТЕМПЕРАТУРОЙ, ВЛАЖНОСТЬЮ ИЛИ АНАЛОГОВЫМ ВЫХОДОМ ВАМ ОБЯЗАТЕЛЬНО НУЖНО САМОСТОЯТЕЛЬНО ПЕРЕЗАГРУЗИТЬ ESP**

augusev, 2019-01-19:
Спасибо! Понятно.