tretyakovsa/Sonoff_WiFi_switch #142: Предложение... Добавить снятие показаний  счетчика расхода воды

Egor2101, 2019-01-20:
С датчика выходит два контакта. При использовании (n) литров воды они замыкаются ( Импульс ). Нужно посчитать расход воды  за месяц.  Как вариант - график расхода в день на протяжении месяца.

renat2985, 2019-01-20:
Данные с графиков не сохраняются. В ESP сохранять эти данные тоже не совсем правильно. Была идея использовать USB разъем в роутере. Большинство новых роутеров его имеют и позволяют реализовать ftp или samba на нем. Кто-то знает красивые компактные решения для ESP позволяющее писать в USB роутера? Это позволит сделать большое хранилище в вашей локальной сети, откроет много возможностей.

А пока возможно вам это поможет: https://github.com/tretyakovsa/Sonoff_WiFi_switch/issues/118
Таким образом вы можете сохранять каждый литр, с временем и датой в excel файл. А потом уже опираясь на эти данные можно самостоятельно создавать графики.

antkachenko, 2019-01-21:
Управляющие компании обычно не используют эту возможность потому, что считывающее оборудование может пропускать этот импульс. А можно как-то гарантировать, что импульс будет посчитан?

antkachenko, 2019-01-21:
> С датчика выходит два контакта. При использовании (n) литров воды они замыкаются ( Импульс ). Нужно посчитать расход воды за месяц. Как вариант - график расхода в день на протяжении месяца.

А какие параметры импульса? Или это просто краткосрочное замыкание контактов? Как его обработать с помощю ESP?

antkachenko, 2019-01-21:
Есть идея еще лучше: сделать сохранение данных во внешнюю базу MySQL. Сейчас на просторах Интернет полно бесплатных и платных сервисов предоставляющих MySQL хостинг бесплатно.

Egor2101, 2019-01-21:
> > С датчика выходит два контакта. При использовании (n) литров воды они замыкаются ( Импульс ). Нужно посчитать расход воды за месяц. Как вариант - график расхода в день на протяжении месяца.
> 
> А какие параметры импульса? Или это просто краткосрочное замыкание контактов? Как его обработать с помощью ESP?

Просто короткое замыкание контактов.

antkachenko, 2019-01-21:
А чего тему-то закрыли? Что Ренат думает про сохранение данных в SQL базу?

lalayan089, 2019-01-21:
InfluxDb + Grafana

Egor2101, 2019-01-21:
Извините. случайно закрыл....

renat2985, 2019-01-21:
@Forldol совершенно прав. Проблема в том что большинство подобных сервисов сейчас работают только с https, а наш проект отправляет только http.

А оптимизировать что-то для какого то конкретного сервиса не вижу смысла, их много, каждому нравится свой, и нет гарантий что через год, два он закроется. В итоге это будет зря потраченным временем.

Отправлять GET по таймеру каждые пол часа, это 48 таймеров. Неудобно. Появилась идея сделать возможность цикличных таймеров.
Например если пишем: 20:**:00 в таком случае таймер будет запускаться каждую минуту с 20:00:00 до 20:59:00.
Пример: **:00:00 тогда будем запускать каждый час.
Пример: **:**:00 запускаем каждую минуту в 00 секунд.
Пример: **:30:00 запускаем в каждые 30 минут каждого часа.
Как вам?

antkachenko, 2019-01-21:
В принципе, это стандартный cron. А какой use case?

renat2985, 2019-01-21:
В таймерах.

Forldol, 2019-01-21:
@renat2985 идея отличная, я за реализацию :)

renat2985, 2019-01-21:
@lalayan089 например?

antkachenko, 2019-01-23:
Собрал схему с отправкой GET запроса на инкрементацию счетчика. Руками если закорачивать контакты "кнопки" - все работает прекрасно. НО: видимо продолжительность "КЗ" (импульса счетчика) очень большая, а при долговременном "нажатии на кнопку" (TACH) модуль не то виснет, не то отключается. Есть способ борьбы с этим?

tretyakovsa, 2019-01-24:
@antkachenko Пишите настройки кнопки и строки сценария для этого счетчика. И какой у вас счетчик. Посмотрю в чем может быть причина.

tretyakovsa, 2019-01-24:
@antkachenko В сценариях что написано?

antkachenko, 2019-01-24:
Если состояние кнопки 0 = 1
Сделать GET запрос на страницу http://increment_cold_water_counter

Если состояние кнопки 1 = 1
Сделать GET запрос на страницу http://increment_hot_water_counter

antkachenko, 2019-01-31:
Коллеги, доброго времени суток.

Есть новости?

tretyakovsa, 2019-02-01:
@antkachenko Расскажите подробнее когда происходит зависание. Если бы я делал, то в первую очередь не вешал бы датчики на GPIO0 GPIO2. И совершенно не ясно каким сервисом пользуетесь http://increment_hot_water_counter - что реально написано в запросе. Как мне попробовать?

antkachenko, 2019-02-13:
Сергей,

Вешаю сборку с модулем на контакты, чрез какое-то время модуль перестает отвечать. Достаточно отключить контакты модуля от счетчиков - модуль оживает...

> Если бы я делал, то в первую очередь не вешал бы датчики на GPIO0 GPIO2.
Поделитесь, как правильно?

Простой PHP скрипт, который делает запись в mySQL базу:

<?php
$servername = "X.X.X.X";
$database = "db_water_counters"; 
$username = "user";
$password = "password";
$now = time();
// Create connection
$conn = mysqli_connect($servername, $username, $password, $database);
// Check connection
if (!$conn) {
      die("Connection failed: " . mysqli_connect_error());
}
 
echo "Connected successfully<br>";
 
$sql = "INSERT INTO `Hot` (`Date`) VALUES (FROM_UNIXTIME($now))";
if (mysqli_query($conn, $sql)) {
      echo "New record created successfully";
} else {
      echo "Error: " . $sql . "<br>" . mysqli_error($conn);
}
mysqli_close($conn);
?>

antkachenko, 2019-03-20:
Сергей,

Вопрос все еще интересен. Могу рассчитывать на ответ?

tretyakovsa, 2019-03-21:
@antkachenko Может вы мне по скайпу позвоните и решим на  вашем примере?

antkachenko, 2019-03-21:
Отлично! Меня просто Ренат все время на форум отправлял...
    	
    	

    	С уважением,Антон Ткаченко.моб. +7 909 675 81 68e-mail: anton@tkachenki.ru