tretyakovsa/Sonoff_WiFi_switch #143: Редактирование сценариев

multysea, 2019-01-20:
Создал простенький сценарий, если температура меньше  23 - включить реле, если больше 25 - выключить. Но, после снижения температуры до 23 реле включилось, а после повышения до более , чем 25 так и не выключилось. Почему так?  И в связи с этим, хотел подвигать параметры сценария, температуру, и не нашел редактирования сценария. В вашем видео об этом тоже ни слова. Можно ли все же редактировать сценарии или нужно только удалять и создавать новые?

renat2985, 2019-01-21:
После добавления сценария связанного с Тимпературой, Влажностью или Аналоговым выходом нужно перезагружать ESP.

multysea, 2019-01-21:
Естественно, что я перезагружал, и не только софт-ресетом, но и хард. И все равно, при понижении включается, при повышении не выключается. Может ли это быть связано с тем, что на это реле у меня есть еще и кнопка?

Egor2101, 2019-01-21:
У меня стабильная прошивка. На реле кнопка + сценарий. Все работает штатно.

Egor2101, 2019-01-21:
На графике температуры у вас отображаются  три линии?

multysea, 2019-01-21:
Да, три, средняя - движется, в соответствии с изменением температуры..

multysea, 2019-01-21:
А может все же кто-то что-то посоветовать:-)? Кроме перезагрузки модуля, ее я изначально  делал..

renat2985, 2019-01-21:
@multysea скорее всего вы что-то не так подключили. Если у других все работает.

multiminik, 2019-01-21:
> А может все же кто-то что-то посоветовать:-)? Кроме перезагрузки модуля, ее я изначально делал..

У вас пин подключенный к кнопке шунтирован через резистор к земле, либо питанию?

multysea, 2019-01-21:
Физической кнопки нет, только виртуальная, в интерфейсе

Egor2101, 2019-01-28:
Сценарий по влажности и температуре на am2301 работал штатно. Подключил ds18b20 . На главной странице есть график и три линии. Но сценарий молчит... При нажатии кнопки управление график не отображается, а шкала температуры от 0 до 1.

multysea, 2019-02-06:
Большая просьба, все же,  кому-то  из разработчиков ответить в этой теме? Сценарии  не работают! Датчик 18b20, и даже резистор между gpio0 и плюсом, как тут советовали, повесил. Все включено правильно, в трех проводах сложно напетлять. Сценарий не  работает, виртуальные кнопки работают, реле переключаются. В сценарии - если температура меньше 25 - включить реле, если больше 29 - выключить. Прошивка стабильная. Модуль сам по себе не перегружается и в режим flash не входит, температуры с DHT11 и 18b20 и влажность с DHT11 показывает, линий на графике температуры c 18b20 - три.

mikemdr, 2019-02-14:
Решил проверить (чуть изменив исходник Scenary.ino) сценарий по-температуре (исходник - последний стабильный, 8266Core 2.3.0, т.к. на >2.3.0 в данном коде некорректно работает обновление по NTP)

`
void orCommand() {
  String Name =  readArgsString();      // Какой параметр проверяем
  String Condition =  readArgsString(); // Операция
  String Volume =  readArgsString();    // Значение параметра
  String test = getStatus(Name);        // получить текущее значение параметра

    Serial.print(Name);//
    Serial.print("(= ");//
    Serial.print(test);//
    Serial.print(" ) ");//
    Serial.print(Condition);//
    Serial.print(" ");//
    Serial.println(Volume);//

  testCommand(Volume, Condition, test);
}

void testCommand(String Volume, String Condition, String test) {
  // последовательно проверяем параметр на соответствие операции сравнения
  // и поднимаем признак исполнения then
...
  if (Condition == ">") {
    Serial.print("Volume ");//
    Serial.println(Volume);//
    Serial.print("Volume.toFloat() ");//
    Serial.println(Volume.toFloat());//
    Serial.print("test.toFloat() ");//
    Serial.println(test.toFloat());//
    if (Volume.toFloat() < test.toFloat()) thenOk = true;
  }
  if (Condition == "<") {
    if (Volume.toFloat() > test.toFloat()) thenOk = true;
  }
...
`
и вот что на выходе (у меня задано два сценария: ">27.00" "<27.00" )

При повышении температуры;
...
t 26.94
t 26.94
t 27.12			// текущая температура с DS1820
f 1				// Признак, что температура превысила заданную

Tach0(= 0) = 1	//  Сценарий по кнопке
thenOk = 0

temperature1(= 27 ) > 27	// 1-ый Сценарий
Volume 27		// заданная температура
Volume.toFloat() 27.00
test 27		// текущая температура из orCommand()[(test = getStatus(Name))].   А не 27.12!!!!
test.toFloat() 27.00
thenOk = 0	// т.к. сравнивается 27.00>27.00, а не 27.12>27.00 , то команды на выполнение НЕ будет

temperature1(= 27 ) < 27	// 2-ой Сценарий
thenOk = 0	
t 27.50
t 27.81
...

При понижении температуры;
...
t 27.25
t 27.25
t 26.62
f 1
Tach0(= 0) = 1	
thenOk = 0

temperature1(= 26 ) > 27	
Volume 27
Volume.toFloat() 27.00
test 26
test.toFloat() 26.00
thenOk = 0

temperature1(= 26 ) < 27
thenOk = 1
comm= pinout not 1
t 25.19
t 24.44
...

Вывод: попробуйте вместо ">" использовать ">=" !!!

mikemdr, 2019-02-14:
>
> если температура при включении модуля ниже порога, то реле включается, и его нельзя выключить даже кнопкой. 
>
И, т.к. проверка кнопки идёт первой, то сначала подаётся команда на выключение, а через несколько **микросекунд** сценарий по температуре подаёт команду на включение.
Попробуйте поставить Сценарий-По-Кнопке последним.
P.S. Я несколько раз перешивал модуль (с предыдущей версией), думая что модуль глючит, т.к. он при нажатии кнопки выключался и тут-же включался, пока не понял, что если модуль включился по таймеру, то его можно выключить **только** отключив таймер  !!!

mikemdr, 2019-02-14:
Ошибка возникала из-за того, что есть 3 функции:
boolean sendStatus(String Name, String volume) {}
boolean sendStatus(String Name, int volume) {}
boolean sendStatusF(String Name, float volume) {}.
У авторов:
       flag = sendStatus(value, getStatusFloat(value));
а должно быть :
      flag = sendStatusF(value, getStatusFloat(value));
иначе float getStatusFloat(value) интерпретировалась как int и записывалась только целая часть.