tretyakovsa/Sonoff_WiFi_switch #144: Логгирование графиков

multysea, 2019-01-20:
В интерфейсе есть графики температуры, влажности, освещенности. Временной период на них достаточно мал, порядка 30 секунд. Вопрос, возможно ли куда-то сохранять данные с датчиков за период, допустим, несколько суток. Если да - то как это сделать, если нет - то прошу объяснить, зачем эти графики нужны вообще? Как я понимаю они неслабо грузят ESP..

renat2985, 2019-01-21:
Данные с графиков не сохраняются. В ESP сохранять эти данные тоже не совсем правильно. Была идея использовать USB разъем в роутере. Большинство новых роутеров его имеют и позволяют ПРОСТО реализовать ftp или samba на нем. Кто-то знает красивые компактные решения для ESP позволяющее писать в USB роутера? Это позволит сделать большое хранилище в вашей локальной сети, откроет много возможностей.

А пока возможно вам это поможет: https://github.com/tretyakovsa/Sonoff_WiFi_switch/issues/118

multysea, 2019-01-21:
Спасибо за подробный мануал, но на мне кажется что это слишком мощные танцы с бубном ради такой функции, подожду, пока можно будет писать на флешку в роутере).