tretyakovsa/Sonoff_WiFi_switch #148: Configs

Egor2101, 2019-01-27:
Здравствуйте Ренат. В меню разработчика есть пункт "Включить порт UART" Как через uart можно взаимодействовать с устройством? Или он используется исключительно для отладки?

kostyamat, 2019-05-28:
Предлагаю вешать на UART расширитель портов, типа такого https://youtu.be/tgV1Nnjo6u8

reo1978, 2019-07-15:
> В основном да. Если придумаете, что туда подключить можно добавить код.
 Сергей можно как то  добавить код о взаимодействии панели Nextion и ESP через UART т.е. на панели Nextion  в программе Nextion Editor создаются элементы например кнопки... каждому из которых присваивается свой id   и этими элементами можно управлять gpio esp -шки а так же передавать данные датчиков на, панель .Хотелось бы использовать эту панель в качестве интерактивного выключателя (панели управления в умном доме) с сенсорным экраном